robotology/mesh-support #781: pwm sign change in head motor control

arrenglover, 2019-03-04:
Currently in the robot configurations, the sign of the gain in the file `iCubGenova02/hardware/motorControl/head-eb20-j0_1-mc.xml` has been changed after the head was re-installed. After talking with @valegagge these parameters shouldn't need to change, instead the motors/encoders/connectors should be mounted in reverse. However, @julijenv has checked that everything in the head seems correctly mounted and connected. 

```
            <param name="pos_kp">                 -300        +300        </param>       
            <param name="pos_kd">                 -10.0       +10.0       </param>     
            <param name="pos_ki">                 -100.0      +100        </param>     
```

```     
            <param name="pos_kp">                 300        -300        </param>       
            <param name="pos_kd">                 10.0       -10.0       </param>     
            <param name="pos_ki">                 100.0      -100        </param>          
```

julijenv, 2019-03-04:
I agree with @valegagge , the PID shouldnt change in between the time it was working before having proceeded with the replacement of the head structure and now. BUT unfortunately, I double check the hardware and electronical wiring and everything is as it commonly expected to be found on a brand new robot head. @valegagge Could you please double check that nothing could have been affected in the config files, let us know

valegagge, 2019-03-05:
@julijenv  ok! I'll check again!

valegagge, 2019-03-20:
Hi @julijenv , 
I checked again the configuration as you required, but there is no changes that justify the PID sign inversion.

julijenv, 2019-03-20:
We will have to check together

julijenv, 2019-03-26:
It has been confirmed that the motor cabling is FAULTY, IT would be changed any time the head come bac for an update, because there is no way to modified the cabling withour umountig hte head, I decicde to leave this issue opend to remind me about it.

Yeshasvitvs, 2019-03-27:
@julijenv please try to fix it by tomorrow morning as we scheduled an experiment for tomorrow. Thanks.

arrenglover, 2019-03-28:
I reverted the PID signs yesterday as we needed to use the robot.

arrenglover, 2019-04-24:
@julijenv do you know if you fixed the head cabling? I thought you had, and that we should have reverted the pwm signs to the standard (i.e. devel branch). However on devel branch the signs of the pwm are incorrect.

Yeshasvitvs, 2019-04-24:
@arrenglover We had a similar problem during our testing with the head (using StandUpWithGreeny branch of robots-configuration) going to hardware fault. @julijenv  suggested to use the changes in head motor control file. They are currently not committed to any branche of robots-configuration.

closed 2019-10-12 by pattacini