robotology/mesh-support #784: Right camera does not work

frank-foerster, 2019-03-06:
### Description of the failure
The right iCub camera stopped to work. 

### Detailed conditions and logs of the failure
As I have not used the robot in a while and did not use the cameras prior to that this may be a fault that has been present for a long time. 
I started both cameras with the yarpmanager using the default script `cameras.xml`, and the default configuration files `dragonfly2_config_right.ini` and `dragonfly2_config_left.ini`. 

The left camera is working fine, only the right one is affected - the displayed image on the yarpviewer is complete black, but the frame of the viewer shows varying frames per second. If this datum is trustworthy it seems that something is coming in, but that something are exclusively black images.

After having read the explanations of  issue #234, I suspect that our camera cable might be dodgy.
I moved the head and eyes to see whether I could get any `flicker' or something the like, and when moving the eyes form left to right, I noticed some small artifacts on the yarpviewer sometimes (but nothing coming close to an actual image). All other head or eye movements did not generate any such artifacts. 

So I just wanted to ask whether there is any other test I can do, or any other thing I can try in order to exclude any potential other error prior to ordering a new cable and replacing the old one.

Many thanks

Frank

julijenv, 2019-03-08:
Hi @frank-foerster ,
from what you explained me, it looks quite clear that the problem is the flat ribbon cable.
You could try to unconnect the the flat cable working to put it on the other caemra and double check that it works.
We bought them from the camera supplier FLIR ref. ACC-01-9000, have a look.
you may find some cable the same kind on RS or FArnell (FFC - 152mm - 0.5mm pitch).
let us know,

frank-foerster, 2019-03-09:
Hi @julijenv,
thanks for the tip. I saw the item number from RS (same here in RS UK) in the other issue (#234) , so I know which cable to get. 
At the moment I don't have time to swap the cable - not sure how to get to it exactly. I'll run with one camera for the time being and might get to this issue once my current experiment is over. 
I will contact you again once I have time to look into this again. 
Thanks for your help.

maggia80, 2019-10-02:
@frank-foerster I'm going to close it. fill free to open it again in case you didn't manage to solve it

closed 2019-10-02 by maggia80