robotology/mesh-support #786: update event-driven FPGA

arrenglover, 2019-03-11:
The FPGA boards of the purple icub relating to event-driven sensors need to be updated / flashed. There are three boards, the zynqBoard in the iCub head and two boards in the iCub eyes. We will need to open the head, un-mount the boards such that they can be updated, therefore I would like to ask for iCub support in opening the head and to coordinate with @mcasti-iit for a time when the FPGA can be flashed. If possible, as early as possible this week would be great! @ChiaraBartolozzi

mcasti-iit, 2019-03-11:
@arrenglover may you provide the MAC address of ZCB mounted in the head?

arrenglover, 2019-03-12:
```
eth0      Link encap:Ethernet  HWaddr 00:0a:35:00:01:22  
          inet addr:10.0.0.3  Bcast:10.0.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1734 errors:0 dropped:0 overruns:0 frame:0
          TX packets:935 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:105309 (102.8 KiB)  TX bytes:71649 (69.9 KiB)
          Interrupt:146 Base address:0xb000 
```

mcasti-iit, 2019-03-15:
On  yesterday afternoon Purple iCub had bot eyes and ZCB updated.
It required more time than expected, due to the lack of an hardware fix (already defined as "tappullo") on left eye. 
Let me know if everything works well. 
I know you'll have a demo this morning, so -in case of malfunctions- I'll re-install the old version.

arrenglover, 2019-03-15:
In the end the upgrade was complete and the cameras are working with the 24bit codec. However, it was a very close call to have everything complete before the demo on Friday morning. I would suggest that in the future, so we give ourselves more time to have things checked before a demo, we:
1. use the open issue to communicate about the work so everyone is up-to-date, and can mention any possible problems.
2. write some documentation for the correct process to upgrade the zynq boards (flashing FPGA, updating uImage, device tree, driver.ko, event-driven software). I guess where to look for the correct version of everything (as suggested by @ChiaraBartolozzi)

Thanks for the effort in any-case @mcasti-iit @julijenv

closed 2019-03-15 by arrenglover