robotology/mesh-support #787: iKinGazectrl overshooting producing bad behaviors iCubGenova03

gonzalezJohnas, 2019-03-11:
### Description of the failure
iKinGazectrl produce overshooting behaviors which results in  values of the encoders outside of the joint limit.

### Detailed conditions and logs of the failure
Tested with the red ball demo and other demos that use iKinGazeCtrl, when the fixation point is to far away (azimuth especially) the robot go there but then remain stuck.

On the yarpmotorgui it can be seen that the encoders values for example of the joint 3 of the head are outside the joint limit.

julijenv, 2019-03-11:
@gonzalezJohnas ,
could you upload a video please?!?!
Could you remove the face cover and try manually to move the joint 3 (eyes I suppose). a video for this manual check could also be useful BUT if you look and see any loose screws let me know.

gonzalezJohnas, 2019-03-12:
I attached the videos (zip file) of the weird behaviors, and I mistaken it was the joint 2 (neck yaw) that produce the error. We checked loose screws and didn't find anything.

[videos.zip](https://github.com/robotology/icub-support/files/2956945/videos.zip)

julijenv, 2019-03-12:
Hi @gonzalezJohnas ,
please ask D. Dellepiane for some loose wire for the AEA encoder of this joint. asj him to follow the sensor above this motor.

closed 2019-03-20 by julijenv