robotology/mesh-support #788: Right arm wrist joints stop responding to yarpmotogui commands

gabrielenava, 2019-03-11:
While the robot was position controlled, still, hanging on its support and in the `home` position, suddenly the right hand wrist joints stop to respond to `yarpmotorgui` commands. I checked the `yarplogger` but there were no errors. However, I anyways dumped the logger which is attached to this issue. 

Then, I entered the `icub-head` pc and I run `FirmwareUpdater`: there was actually one board in `idle` state, but I have no idea if this board actually caused the issue. See picture below:

[image]

[log_icub-head_yarprobotinterface_1375.txt](https://github.com/robotology/icub-support/files/2953152/log_icub-head_yarprobotinterface_1375.txt)

julijenv, 2019-03-11:
Hi @gabrielenava , from the FirmwareUpdater unfortunately the problem is more complicated than you suggested.
the idle board is for sure not related to the right arm fault, because the ems4 ID 10.0.0.1.10 is related to skin for the left leg.
I cant find any error from the log file though.
give it another try and see it doesnt start.

gabrielenava, 2019-03-13:
It happened again, and again I do not have any error or information concerning the problem. However, thanks to @Giulero we managed to shoot a video of the issue. In particular, the robot is performing the `Yoga ++` demo, and at a certain point the right hand wrist joints stop being position controlled. `THERE IS NO HARDWARE FAULT` on that joint. Note that then the "communication is restored" (I actually don't know what is going on) and the joints go back under control. `This is not required` by the Simulink controller as the torque control stops at the robot elbow (therefore we are not asking the wrist joints to move).

[image]

julijenv, 2019-03-13:
hi @gabrielenava ,
I will have a look ASAP I can ,I guess one of the wrist motor cable is pinched during the extension of the elbow.

julijenv, 2019-03-18:
Logically it has been solved , did you notice the same behaviour @gabrielenava ?
if it is gnone please close the issue, thx

gabrielenava, 2019-03-18:
No, it seems now the problem disappeared. Thanks!

closed 2019-03-18 by gabrielenava