robotology/mesh-support #789: Left Arm Capacitor burnt

pencoj1, 2019-03-12:
### Description of the failure

We got a spark on the iCub. I don't know where it came from because we just saw this capacitor jumping out the robot and now we cannot control the left arm.
[image]

preliminary investigation with Daniele Pucci and Marco Maggiali said it is a " condensatore al tantalio che ha fatto boom" 
@maggia80 

### Detailed conditions and logs of the failure

<#STRING_START#> from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 56s 982m 392u: (code 0x00000013, par16 0x0029 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
<#STRING_END#>
1805
16,350585
5
<#STRING_START#> from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src CAN1, adr 0, time 56s 930m 116u: (code 0x00000019, par16 0x3f01 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
<#STRING_END#>
1806
16,350631
4
<#STRING_START#> from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 56s 930m 255u: (code 0x0000000b, par16 0x022d par64 0x0140015603b3013f) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .

julijenv, 2019-03-13:
Hi @pencoj1 ,
Could you open the left upper arm cover and investigate inside it. for sure it is coming from one of the MC4 board. I guess one of the wire from the power connector got broken or melt, could you double check this or make pictures ?!?
Let us know,

pencoj1, 2019-03-13:
[image]

Where are the MC4 boards located? could you send me a sketch pls?

julijenv, 2019-03-13:
Hi @pencoj1 ,
please catch me on skype this safternoon around and later to 3ppm my skype Id is icub.support.
C U this afternoon

closed 2019-03-26 by pencoj1