robotology/mesh-support #791: Shoulder tendon is loose

tasbolat1, 2019-03-15:
### Description of the failure
After having experiment involving right arm, the right arm becomes shaky in some configurations. I also noticed that the one of the tendon is loose. How can make it tight again?

This is a picture where I can see that the tendon is loose.

[image]

### Detailed conditions and logs of the failure
There is no error or hardware failure in the log file.

julijenv, 2019-03-15:
Hi @tasbolat1 ,
if you go to the icub wikipage [Tendons_and_replacement](http://wiki.icub.org/wiki/Tendons_and_replacement) and download the document ShoulderWiring you will be able to see that this cable is shoudler cable number 6 which is antagonist with 5 and to tighten it you need to tighten both 5 and 6 tensionner you can find on this picture
[image]
Let me know,

tasbolat1, 2019-03-18:
Hi @julijenv,
I have managed to tighten the tendons. So far seems good. Yet need to see the results in the real experiments. Thanks a lot.

closed 2019-03-18 by tasbolat1