robotology/mesh-support #793: No icub/inertial when starting yarprobotinterface

stu105502, 2019-03-20:
### Description of the failure
@towardthesea, @julijenv 

Hello @julijenv,

you know me already from issue #740 and I'm working with the iCubHamburg01. In this issue we had some problems related our robot because of some defect robot-configuration files (a spelling mistake). This is not related to our current problem, but you wanted to take care that the error is being fixed and that our files are being upgraded to motioncontrol v5. Could you please try to remember the responsible persons again? This could be later a serious issue for us again.

To our new problem: We noticed that our head now properly starts with the yarprobotinterface, but the service "icub/iertial" is simply not started (as far as I understand the Service regarding the Inertial Sensor does not start). After a little searching I probably found the problem. I assume that this service should start with yarprobotinterface and that there is another mistake that was made in our robot-configuration files. In icub_all.xml the line that should probably include the /icub/inertia service is simply commented. Even uncommenting the line does result to an error, because we can not access the subdevice "imuBosch_BNO055" and no plugin was found.

I tried to rebuild yarp, icub and co, but this brought no results. My assumption is that we either use another subport or that the subdevice was changed, when we sent the iCub for an upgrade 2 years ago. Our head is basically a HeadV1 that was upgraded to a HeadV2, but we still have some unique characteristics (ETH boards insted of cfw2can). Could you maybe give me a hint, how I can check the subdevice or the port.

julijenv, 2019-03-20:
Hi @randaz81 , @valegagge , @marcoaccame ,
Could you please double check their config files, on the issue #740 there was a spelling mistake, this time it looks like another problem could have been inserted by mistake.

julijenv, 2019-03-20:
Hi @stu105502 , 
from this picture of Head V1 (your is V2 but the boards level should looks like this picture)
[image]
Could please tell me if you have this device
[image]

or this one : 
[image]

if you are in the first case please replace the head-inertial.xml file inside hardware/inertial with this following file.

[head-inertial.txt](https://github.com/robotology/icub-support/files/2988380/head-inertial.txt)
change the .txt in .xml (I could nt send otherwise to compress it).
Please enable xsens when you compile icub.
thx @davidetome ,
let us know

stu105502, 2019-03-20:
Of course, I'll look into it immediatedly tommorrow. If you are interested, I could also send you the modified files.

valegagge, 2019-03-20:
>Could you please double check their config files, ....

Are the current configuration on  master branch?

stu105502, 2019-03-21:
@valegagge,
which config files are you reffering to? The incorrect files are still in the git-main but the files we have configured doesn't seem to be in the git yet. We received some files from @julijenv whose purpose it was to upgrade our robot-configuration files from motion-control v3 to v5.

Furthermore, we changed some files manually because the files in the git are icorrect. in "iCubHamburg01/calibrators/head_calib.xml" we have in line 5 a spelling mistake head-calibartor instead of head-calibrator. Now we probably found in icub_all.xml the problem that the line regarding the inertia sensor is simply commented (line 12).

Since our last issue, our robots-configurations were not updated and the defect files are still in the git

julijenv, 2019-03-21:
Hi @stu105502 ,
ASA you make your config files working properly, just send us your repository and we will update the overall files that are not correct in git.
cheers,

stu105502, 2019-03-21:
Alright thanks, our robot configurations work, when we ignore the commented line in our icub_all.xml, so our modified files do not solve the problem with the Interial Sensor. I'll send it immediately if our Inertial Sensor is running, too. I have watched the hardware in our iCub and we have indeed the blue chip. I'll send you the picture in a second, I tried to start again our yarprobotinterface with the uncommented line and received the messages that I will attach at the end of this post.

The problem seems simply to be that we have not build the "imuBosch_BNO055.ini" as plugin. I took a look on our yarp compilations of the pc104 and found out that ccmake does 
restrict us to set the variable "ENABLE_yarpmod_imuBosch_BNO055" to true because "YARP_HAS_I2C" is needed to build the plugin. 
Could you give me a hint how we can fix this? We run our iCub currently on a live image on an USB with version 7.9-17.12.18.

We are currently working on the 

```
icub@pc104:/usr/local/src/robot/yarp/build-pc104$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/usr/local/src/robot/yarp/build-pc104/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/yarprobotinterface.ini
||| finding file [config]
||| checking [/usr/local/src/robot/yarp/build-pc104/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml 
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml 
[DEBUG]Preprocessor complete in:  0.0179412 s 
||| finding paths [plugins]
||| checking [/usr/local/src/robot/yarp/build-pc104/plugins] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/plugins] (robot)
||| checking [/home/icub/.config/yarp/plugins] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/plugins] (YARP_DATA_HOME)
||| checking [/etc/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins
||| checking [/usr/local/src/robot/icub-tests/suits/plugins] (YARP_DATA_DIRS)
yarp: Port /icub/yarprobotinterface active at tcp://10.0.0.2:10002/
[INFO]startup phase starting... 
||| finding paths [plugins]
[DEBUG]eth::parser::print(pc104Data) for PC104: 
[DEBUG]PC104/PC104IpAddress:PC104IpPort =  10.0.1.104:12345 
[DEBUG]PC104/PC104TXrate =  1 
[DEBUG]PC104/PC104RXrate =  5 
[DEBUG]EthSender is a PeriodicThread with txrate = 1 ms 
[DEBUG]EthReceiver is a PeriodicThread with rxrate = 5 ms 
[WARNING]in EthReceiver::config() the config socket has queue size =  2097152 ; you request ETHRECEIVER_BUFFER_SIZE=  
[DEBUG]eth::parser::print(boardData) for BOARD head-eb20-j0_1 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES: 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.20 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  mc4plus 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS: 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/Name =  head-eb20-j0_1 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60 
[DEBUG]eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.20 (head-eb20-j0_1) with timeout = 0.02 sec and period of missing report = 60 sec 
[DEBUG]TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type mc4plus with IP =  10.0.1.20 
[WARNING]embObjMC BOARD  head-eb20-j0_1 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails) 
[INFO] from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 377s 123m 6u: (code 0x00000034, par16 0x0001 par64 0x0000000000000014) -> SYS: the board has detected a restart of the remote transceiver because the rx ropframe sequence number is 1. In par64 there is the expected number. + .
[DEBUG]EthResource::verifyBoardPresence() found BOARD head-eb20-j0_1 with IP 10.0.1.20 after 0.00364184 seconds 
[DEBUG]EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD head-eb20-j0_1 with IP 10.0.1.20 
[DEBUG]EthResource::cleanBoardBehaviour() has cleaned the application in BOARD head-eb20-j0_1 with IP 10.0.1.20 : config mode + cleared all its regulars 
[... cut ...]
[ERROR]head-mc_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device head-mc_wrapper 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
[WARNING]There was some problem running actions for shutdown phase . Please check the log and your configuration 
[INFO]shutdown phase finished. 
[ERROR]Error in shutdown phase... see previous messages for more info 
[INFO]RFModule failed to open.
```

julijenv, 2019-03-21:
Hi @stu105502 ,
could you please copy and paste the result of this command on a shell from the pc104, please?
cat /VERSION_*
thx in advacne

stu105502, 2019-03-22:
Of cource @julijenv,
here is the output

```
iCub LIVE image for PC104 version 7.9-17.12.18
Based on stretch amd64
Build on 18-Dec-2017
```

stu105502, 2019-03-24:
In Yarp "ENABLE_yarpmod_imuBosch_BNO055" is only restricted because of "YARP_HAS_I2C". As far as I understand our imuBosch device shall be connected with an serial-usb-port. Neither our robot-configurations or folder /dev seems to provide any signs of i2c like it is in #763.

Could you explain to me what the the condition is to have "YARP_HAS_I2C" as true or if the plugin should be built by compiling another repository?

julijenv, 2019-03-26:
Hi @stu105502 ,
I do believe that your robot as actually an inertial sensor that is connected to an RS232/USB dongle so actually you dont need to use I2C protocol. in this case you head-inertial.xml was correct from the beginning, could you double check and let me know, thx in advacne

stu105502, 2019-03-29:
We seem to have an USB converter at the position. At least this is what I would assume from following result 

```
icub@pc104:/usr/local/src/robot/robots-configuration/iCubHamburg01/hardware/inertials$ sudo dmesg | grep tty
[    0.000000] console [tty0] enabled
[    0.914470] 0000:01:00.0: ttyS0 at MMIO 0xf7b00000 (irq = 16, base_baud = 7812500) is a XR17V35X
[    0.914624] 0000:01:00.0: ttyS1 at MMIO 0xf7b00400 (irq = 16, base_baud = 7812500) is a XR17V35X
[    0.914770] 0000:01:00.0: ttyS2 at MMIO 0xf7b00800 (irq = 16, base_baud = 7812500) is a XR17V35X
[    0.914917] 0000:01:00.0: ttyS3 at MMIO 0xf7b00c00 (irq = 16, base_baud = 7812500) is a XR17V35X
[   14.830600] cdc_acm 2-7:1.0: ttyACM0: USB ACM device
[   14.938208] usb 2-8: FTDI USB Serial Device converter now attached to ttyUSB0

```

I have also double checked the head-inertial.xml. It seems everything in place. Now is only the question why the plugin or the .ini for the ImuBosch is not built properly. Are these built by compiling the robot-configuration?

julijenv, 2019-04-01:
Hi @stu105502 ,
Ok let's try to find out where hold the problem. From what I could understand the Config files should be OK. The device is detected (from your last post). You created a new image pc104, right?
which one did you take, from which icub wikipage?
which steps did you follow to compile both server side and pc104?
Did you first update every repositories? (YARP, iCub, etc..)

I see from the log file reported above
> [ERROR]Cannot find "imuBosch_BNO055" plugin (not built in, and no .ini file found for it)
Check that YARP_DATA_DIRS leads to at least one directory with plugins/imuBosch_BNO055.ini or share/yarp/plugins/imuBosch_BNO055.ini in it
yarpdev: ***ERROR*** could not find device <imuBosch_BNO055>
[ERROR]cannot create device <imuBosch_BNO055>
[ERROR]ControlBoardWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <inertial> was found but could not open

can you try to find the plugins/imuBosch_BNO055.ini inside the head.
Did you enable the line imubosch in YARP compilation (inside ccmake options)?
Aas previously, explained you should have imubosch attached to USB dongle so actually your head-inertial should looks ike this.

[head-inertial.txt](https://github.com/robotology/icub-support/files/3028784/head-inertial.txt)
please change .txt into .xml

Let me know,

stu105502, 2019-04-02:
@julijenv,

For installation on the pc104 laptop, I followed the instructions given on [this side](http://wiki.icub.org/wiki/The_Linux_on_the_pc104), which led to a [zip-archive](http://www.icub.org/download/pc104-images/) where I downloaded the most recent  live-image 7.9 from the 18.12.2017. I am properly able to access the pc104 and (when I exclude the inertial Sensor), I'm very well able to run and start the applications from the yarpmanager. I doubt that there is a problem of communication.

For the compilation: When I set up the new live image I updated every git repository and I recompiled everything multiple times regarding [the instructions for the pc104](http://wiki.icub.org/wiki/Compilation_on_the_pc104) and the [instructions for sources](http://wiki.icub.org/wiki/Windows:_installation_from_sources).

For the plugin: I'm sorry. I seem not have make this clear. The point is that we find the ini-file (see attachement for console output) for the imuBosch device in yarp, but not in our building-workspace We can not activate the build of the device because it is told that we need the variable "YARP_HAS_i2C" as true for building it in the first place (see picture in attachement). Although we use a usb dongle, i2C seems to be essential for building the ini in the first place.

```
icub@pc104:/usr/local/src/robot/yarp/build-pc104$ find $ROBOT_CODE -name "imuBosch*"
/usr/local/src/robot/yarp/build/src/devices/imuBosch_BNO055
/usr/local/src/robot/yarp/src/devices/imuBosch_BNO055
/usr/local/src/robot/yarp/src/devices/imuBosch_BNO055/imuBosch_BNO055.cpp
/usr/local/src/robot/yarp/src/devices/imuBosch_BNO055/imuBosch_BNO055.ini
/usr/local/src/robot/yarp/src/devices/imuBosch_BNO055/imuBosch_BNO055.h
/usr/local/src/robot/yarp/build-pc104/src/devices/imuBosch_BNO055
```

[image]

stu105502, 2019-04-03:
@julijenv 

hey, we have managed to fix the problem ourself. We found all our problems in the source of the imuBosch device ($ROBOT_CODE/yarp/src/devices/imuBosch_BNO055). There are some issues that gave us quite problems:
1) In the CMakeLists.txt the variable "YARP_HAS_I2c" was a variable that was set as required. Probably you set this variable because there are build three sources that are dependend on I2C. 
2) After we have made a workaround we were able to compile everything but in the building process, we encountered another problem. You guys have in the imuBosch_BNO055.cpp defined functions for as well i2c and serial connections. Our problem was now that we were missing the library "linux/i2c-dev.h" and we failed in having the needed functions given.
3) After installing this library (the library was not interested and should only provide us with the possibility to compile in the first place), we were able to compile everything.
4) But at the startup of yarprobotinterface we received an unknown message type error.

After this we simply exchanged the files of the imuBosch device with the files of the former version (yarp 3.0.1). Now we have access to the device and can start the iKinGazeCtrl for example. I have actually no clue, why the newest version files of yarp of the imuBosch device does not work, but I would suggest that you guys keep the i2c and serial versions in separate files and decide regarding the variable "YARP_HAS_I2C" which of these files you use for the build. 

Furthermore, after we are now sure that our robot configuration seem in check, I would like to send you the files that has to be updated in our git-repository (fixing spelling check, uncommenting the inertial device and the newest motioncontrol version). Could you please provide me with some contact data, where I should send/depose them?

julijenv, 2019-04-04:
Hi @stu105502,
please send me the modified files reveling by doing
> git status
I will commit them later on.

stu105502, 2019-04-05:
@julijenv,

of course, thank you very much! The git status response is in the attachement. Shall I send them directly to you by mail or shall I simply include them into this issue? They are quite a lot because of the motionncontrol upgrade to v5

```
icub@wtmicublaptop:/usr/local/src/robot/robots-configuration$ git status 
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   iCubHamburg01/calibrators/head_calib.xml
	modified:   iCubHamburg01/hardware/mechanicals/head-eb20-j0_1-mec.xml
	modified:   iCubHamburg01/hardware/mechanicals/head-eb21-j2_5-mec.xml
	modified:   iCubHamburg01/hardware/motorControl/head-eb20-j0_1-mc.xml
	modified:   iCubHamburg01/hardware/motorControl/head-eb20-j0_1-mc_service.xml
	modified:   iCubHamburg01/hardware/motorControl/head-eb21-j2_5-mc.xml
	modified:   iCubHamburg01/hardware/motorControl/head-eb21-j2_5-mc_service.xml
	modified:   iCubHamburg01/icub_all.xml
```

julijenv, 2019-04-05:
send them to me please
> julien.jenvrin@iit.it

thx

julijenv, 2019-04-08:
Hi @stu105502 ,
to close the issue I need to receive your updated files in order to push them on your repository.
thx in advance.

julijenv, 2019-04-24:
Any news??? @stu105502 @towardthesea . I need your last sonfiguration to push them on the maste repository, thxi advance

closed 2019-10-02 by maggia80