robotology/mesh-support #794: iCub no even pinging

TheCamusean, 2019-03-22:
### Description of the failure
Hello everyone,

We have tried connecting through ssh to the iCub 
`ssh pc104`

but it was not connecting, so we connect an screen to the iCub's pc and the following was appearing : 

> Probing  EDD (edd=off to disable)... ok

> XZ-compressed data is corrupt

>  -- System halted_

julijenv, 2019-03-22:
Hi @TheCamusean ,
I guess that the USB key is corrupted, you may need to recreate a USB key.
BTW, which robot are u using?
Let us know.

TheCamusean, 2019-03-22:
Hello, 

I am using iCubDarmstadt01. 

When you say the the USB key is corrupted, what do you exactly mean? That the operative system is corrupted ? Can you give me a guideline to recreate the USB key? 

Best wishes

julijenv, 2019-03-22:
Hi @TheCamusean ,
Please follow this webpage from iCub wiki: [pc104](http://wiki.icub.org/wiki/The_Linux_on_the_pc104)
please do it on another usb key compatible, just in case.
Let us know

stu105502, 2019-03-24:
Hey @TheCamusean,

I'm just writing this because we have also encountered the problem two months ago. We noticed that our iCub used as default network for server communication the port 10.0.1.104 to communicate with the server laptop. This network was never thought for communication with the server-laptop but seems to be an internal network to access different boards built in your iCub.

You could check, if you have the same problem by entering the network with an ip in this network (For example 10.0.10.1) and pinging the address 10.0.1.104 (pc104/iCub). If you receive an answer you know that the iCub simply uses the wrong network. You can see the network configuration file at this [page](/etc/network/interfaces) under network configurations. This could be more helpful to understand what I mean.

How we personally solved this: The problem is that eth0 seems to be used by the iCub as default network for communicating with server but you are interested to use eth1 as default. You can solve this by simply exchanging the configurations of these two interfaces. You can achieve this by:
1) Enter the iCub by either ssh over the 10.0.1.255 network or by working with a keyboard and monitor directly connected to the pc104.
2) take a look at the iCub file _/etc/network/interfaces_.
3) Just copy the the content in eth0 to eth1 and vice versa.
4) Restart the ICub and try to ping the iCub.
5) If this does not work you maybe have to replay the usb with a new live version, but this was actually the way how it worked for us personally.

TheCamusean, 2019-03-25:
@stu105502 Thanks for the recomendation, but in our case, the OS was corrupted and was not working.

@julijenv I am finding an awkward situation. I have been able to install the OS in the pc104, I changed also the ROBOT_NAME in the .bashrc_iCub. Now, I am able to connect passwordless through the laptop via ssh pc104. 
Anyway, when I am running the yarpmanager in the laptop I am having troubles. I go to Cluster and there  I click run in the Nameserver node box. When I do this, the terminal is asking me to enter the icub@icubsrv's password, that actually, we forgot.

1. Do you know what is the generic password in the icubsrv laptop?
2. Before I did not need to enter the password of when I was clicking run on the cluster ->Nameserver node, is there a way to avoid it?

Thanks alot for the support

TheCamusean, 2019-03-25:
@julijenv Is there any link to set the latest version of the icub server in any workstation?

julijenv, 2019-03-26:
Hi @TheCamusean ,

> 1. Do you know what is the generic password in the icubsrv laptop?

In theory it is 

> icub or icubAdmin

For this instead
> 2. Before I did not need to enter the password of when I was clicking run on the cluster ->Nameserver node, is there a way to avoid it?

You actually need to create a ssh key in between both pc104 and icubsrv and icubsrv and pc104 in order to have them reckonize one to another.
follow these steps from [here](http://wiki.icub.org/wiki/The_Linux_on_the_pc104)

For this matter ;

> Is there any link to set the latest version of the icub server in any workstation?

The overall wiki has been recreated with the support on 16.04 ubuntu LTS but I know that it is possible to make work with 18.04 LTS, give it a try.

Let us know,

TheCamusean, 2019-04-04:
Hello @julijenv ,

I am going to explain carefully all the steps I followed and then maybe you can understand what I miss doing. The actual state is that I can ssh or ping the pc104, but when I am running yarpmanager and trying to get connection with the iCub through the cluster module, the pc104 is not connecting.

The steps I followed are these : 
1. I installed the last version of the OS in the icub, the ip for the icub is 10.0.0.2 and changed the name of the robot YARP_ROBOT_NAME to iCubDarmstadt01 in the iCub_bashrc
2. I eliminated the known hosts in my laptop and created a new ssh key in the laptop to connect with the pc104 passwordless.
3. When running yarpmanager, in the cluster module, the part related with the icubServer is getting green, but the pc104 remains red. 

Am I missing any step in the installation procedure? Do you have any suggestion?

Thanks alot

julijenv, 2019-04-04:
Hi @TheCamusean ,
if you have some spare time we could investigate your problem over a teamviewer-skype meeting?
my skypeID is icub.support
let me know

TheCamusean, 2019-04-05:
Sure,

I write you there in order to fix a meeting

julijenv, 2019-04-05:
there was some issues with the ssh mirroring to reckonize pc104 to icubsrv and icubsrv to icubsrv.
There was also problem with the namespace.
everything got cleared

closed 2019-04-05 by julijenv