robotology/mesh-support #798: Firmware hands iCubNancy01

pencoj1, 2019-04-02:
Our iCub has the version 1 for the hands. With @julijenv we where checking that when we manually move the fingers, the value of the encoder in yarpmotorgui does not change.

We were wondering if the firmware of our hands is the correct one
[image]

julijenv, 2019-04-03:
Hi @marcoaccame ,
Does it ring you a bell? On yarpmotorgui all the values from the hand keep being at zero. It looks like the same issue that we saw quite recently but I dont remember what was the source of the non transmitting values to the yarpmotorgui? Was it the EMS or the MAIS?!?
Do you remember?
let me know, thx in advance.

marcoaccame, 2019-04-03:
The values are not broadcasted by the ems boards at 10.0.1.2 and 10.0.1.4 because we must tell them to broadcast also the position values.

Or better, we must tell them in the correct way. The xml files have a wrong syntax.

You must edit the files `left_arm-eb2-j4_15-mc_service.xml` and `right_arm-eb4-j4_15-mc_service.xml` and correct a line.

```
errata:
<group name="BROADCASTPOLICY">
    <param name="enable">    eomc_mc4broadcast_position    status  print   pidvalues     </param>
</group>

corrige:
<group name="BROADCASTPOLICY">
    <param name="enable">    position    status  print   pidvalues     </param>
</group>

```

We should commit that line to master.

julijenv, 2019-04-03:
double chcek with @pencoj1 and everyting is alright

closed 2019-04-03 by julijenv