robotology/mesh-support #800: icub-lyon01 blades dont boot

pfdominey, 2019-04-04:
Greetings,
After the move to Dijon (see photo of new lab) 
http://robotcognitionlab.pagesperso-orange.fr/

we have been doing teleoperation of the head for several weeks with no problem.

Now we will start using ikin for pointing trajectory generation, and I was to boot the blades to run ikin, and strange problem, blades fail to boot.  They do not find a boot device.

I wonder if in the move to dijon there was a cable that did not get connected...

from a reladted issue I saw this

> As an additional precaution, since the CISCO network switch takes a lot of time to boot up, please wait until the light on the connected ports becomes GREEN (about 5 minutes) before powering the icub-server up,  then after other 5 minutes the icub-server is started correctly 
> and you can turn on the blades and the CUDA machine.
> ...Matteo
> https://github.com/robotology/icub-support/issues/41
> 

the pc104 and ikart boot and connect so I think the network switch is ok...

here is a link to some photos
https://photos.app.goo.gl/4cxBiCSv8PNxxet18

thanks for your help!

cheers 
Peter

pattacini, 2019-04-04:
Hi @pfdominey 

Our ICT responsible is on leave now. We will try to collect as much info as possible and get back to you.

pattacini, 2019-04-05:
Hi @pfdominey 

@mbrunettini should be back next week (not sure which day exactly) and I'm afraid we need him for this kind of issue.

However, in the meantime, you may consider running iKin-based modules on the iKart instead. This workaround would let you go forawrd with your experiments.

mbrunettini, 2019-04-08:
If the vserver is not working the blades can't boot..
Unfortunately, inthe pictures of vserver booting you sent me no errors are displayed and it seems that the lower part of the screen is missing.
Can you shot a video or wait until an error is shown on screen (the server BIOS takes a while to boot)?
Thanks.

pfdominey, 2019-04-10:
Hi Matteo,
Thanks.  I will send you this information later today.
In the mean time, I am trying to install the simlator for windows from
binaries.   I have installed Yarp, but I cant seem to find the download for
the binary files fro the simulatior.  Can you end me alink and tell me if
there is anything speial I need to do?
thanks
cheers
Peter

mbrunettini, 2019-04-10:
I suggest you to open another issue for the simulator probelm, so we can assign to the appropriate expert and so give a better support. Thanks

pfdominey, 2019-04-10:
greetings,
Here is the video of the failure to boot for the icub-vserver
https://photos.app.goo.gl/8mM6wBKr6XCtXyZm6
thanks
cheers
Peter

pfdominey, 2019-04-19:
Greetings,
Any advice?
thanks

mbrunettini, 2019-04-23:
The system is not able to load the vmWare OS, this could be a problem with the storage were the OS in is installed or a misconfiguration in the BIOS: **check in the BIOS if the system is booting from the correct resource.**
I think your system boots from the SD card, but if you send me the DELL SERIAL number we can be sure.
Matteo

closed 2019-10-12 by pattacini