robotology/mesh-support #802: PC104 update

Tobias-Fischer, 2019-04-04:
Hi @julijenv,
iCubLondon01 is using a (very) old version of the PC104 with a 32-bit processor. Recent pc104 images only support the 64-bit processors (http://www.icub.org/download/pc104-images/). Hence we cannot update the pc104 image.
We'd like to get information whether it is feasible to install a new PC104 ourselves, or whether we need to ship our robot to IIT.

Best, Tobi and @umgnunes

julijenv, 2019-04-04:
Hi @Tobias-Fischer ,
we have to ge tin touch @maggia80 , to undertand what is the best solutuion for the upgrade of your head. I will go back to you ASAP

Tobias-Fischer, 2019-04-04:
Thanks @julijenv!

julijenv, 2019-04-05:
@maggia80

maggia80, 2019-04-08:
Dear @Tobias-Fischer, 

The PC104 with 64-bit processor that we were using on our iCub is obsolete. We moved to a new standard that is the COM-Express Type 6. However, I am searching for a PC104 board as an alternative. I'll let you know if I found something suitable

Tobias-Fischer, 2019-04-08:
Hi @maggia80,
Thanks for letting us know! Is it possible to use the new board (COM-Express) on iCubLondon01? I was not aware that the PC104 is obsolete. I don't think it is a good idea to move from one obsolete board (32-bit PC104) to the next obsolete board (64-bit PC104) ..

Best, Tobi

maggia80, 2019-04-09:
hi @Tobias-Fischer , unfortunately, the COM-Express cannot be mounted on iCubLondon01, because the COM-Express cannot be connected to the CFW board (the board with the 10 CANbus lines). We use the COM-Express for the robot with the Ethernet cards. We should substitute almost the boards of your robot and some mechanical parts.

Tobias-Fischer, 2019-04-09:
Hi @maggia80,
Ahhh, I understand. In that case, a 64-bit PC104 might be better indeed. Please let me know if you find one.

Best, Tobi

maggia80, 2019-04-23:
Hi Tobias,

I have found a candidate alternative to the old PC104. It has not very high performance, but it might be enough. The board is the CM2-BT2-E3825-ETT [https://www.adlinktech.com/Products/DownloadMDownload.aspx?lang=en%20%20%20%20%20&pdNo=1353&kind=D](url). We didn't test it and we are still deciding whether or not we go for testing it. In case you wanna try here the list of all the components and a possible supplier: 
[Carrello_apr23_0429.pdf](https://github.com/robotology/icub-support/files/3106885/Carrello_apr23_0429.pdf)

Tobias-Fischer, 2019-04-25:
Hi @maggia80,
Thanks for checking this out! I think it looks good - it's not too expensive and should last at least a little bit longer. It would be great if you could give it a try so we know whether we can use it on iCubLondon01.

Best, Tobi

maggia80, 2019-04-29:
Hi @Tobias-Fischer, we are going to buy one and test it. I'll let you informed

Tobias-Fischer, 2019-04-29:
Great thanks a lot!

Tobias-Fischer, 2019-05-30:
Hi @maggia80, did you have a chance to test the new PC104? Would be great to hear how things are going. Best, Tobi

maggia80, 2019-06-07:
Hi Tobias, we have just received the board. We tested the USB image and it works, but we still need to test the CFW-002 board, CAN and Firewire for the cameras.

Tobias-Fischer, 2019-06-07:
Sounds good, thanks @maggia80!

Tobias-Fischer, 2019-07-11:
Ping @maggia80

maggia80, 2019-07-26:
sorry Tobias,  but we are still working on it. the activity does not have high priority and therefore it takes longer than expected.

Tobias-Fischer, 2019-10-08:
Hi @maggia80,
Did you make any progress on the PC104 update? Unfortunately, due to recent changes in the requirements of YARP/icub-main (C++11 etc.), we basically can't update our system anymore.

Best, Tobi

traversaro, 2019-12-07:
cc  @maggia80

pattacini, 2019-12-09:
Hi @Tobias-Fischer 

We will revive the task this week. I'll be checking the status of the board with @mbrunettini and get back to you.

Stay tuned 👍

Tobias-Fischer, 2019-12-09:
Thanks @pattacini :)! Please keep @umgnunes in the loop, he'll take care of iCubLondon01 once I move on from London (which will be shortly).
/Cc @yiannisdemiris

pattacini, 2019-12-16:
@Tobias-Fischer @umgnunes @YiannisDemiris a super quick update on this.

We have recently resumed the verifications on the new board. However, we expect to finish up testing at the beginning of January, when we should be able to send it to you with everything needed already installed.

umgnunes, 2019-12-16:
Great! Thank you @pattacini

pattacini, 2020-01-21:
Hi @umgnunes 

This is to let you know that we have completed the tests on the new board.
The system boots correctly with the latest iCubOS image (Ubuntu 18.04); network connections work fine as well.

I think we can provide you an offer then.
cc @maggia80

umgnunes, 2020-01-21:
Hi @pattacini,

Great news! We are interested in updating our pc104 ASAP. Please let me know how proceed.

pattacini, 2020-01-21:
Great then 👍 
Just get in touch with @maggia80 via email: marco.maggiali _at_ iit.it

pattacini, 2020-01-21:
I'd say that this can be closed now.
@umgnunes please proceed as agreed.

cedricgoubard, 2021-07-12:
Hello, 
Sorry to resurrect this thread; I am taking over from @umgnunes on trying to fix our iCub, and we still have the same 32bits pc104.  
Here is the last email we received from your team, back in March 2020:

 ---
Sorry for the very late answer. We tested the PC104 (64 bit version) by its own, but we didn’t install it in our 32bit robot yet. This is because of three reasons:

1) the computational power of this solution is lower than the older PC104.
2) the dimension of the heatsink is a bit different from the older PC104 and we need to check it in the robot
3) the connectors are different from the older PC104 and therefore if we do the change of PC104 we cannot go back to the old solution.

For these reasons I do not suggest you to move to the new PC104. However, we upgraded quite recently the OS of the system and it may solve your issue:

https://github.com/robotology/QA/issues/369
Let me know if it solves your issue, or if we need to prepare you a quotation for changing the PC104.
---

I do not think this was applied to our iCub. Is this solution still relevant? Should I go ahead and try to upgrade our 32bit pc104 with this new distribution, or have any other solution come to light?

I appreciate any guidance you could provide.
Thanks,
Cedric

pattacini, 2021-07-12:
Hi @cedricgoubard 

Let me call for some people in the know here: @maggia80 @mbrunettini @andreaderito

maggia80, 2021-07-12:
@cedricgoubard you are right the proposed pc104 is not really powerful. There is a new possible candidate from the same company:

https://www.adlinktech.com/Products/PC104SBCs/PCI104-Express/CMx-SLx?lang=en

or from **rtd**: https://www.rtd.com/NEW_manuals/hardware/cpumodules/CMA24CR_BDM610000085A.pdf

We didn't test them both. They require some checks in the mechanical specs (heatsink) and connectors. There is a huge difference in the case of the **rtd**, it requires both `5V and stable 12V`.  While CMx-SLx needs only 5V input.

cedricgoubard, 2021-07-12:
Hi @maggia80 
Thank you for answering so quickly! 
How could we make sure that these boards would fit our robot? Are there any checks we could run on our end? Do you need more details about our hardware? Our iCub is quite old, I think it was the 3rd ever made...

maggia80, 2021-07-12:
Hi @cedricgoubard,

the adlinktech board has an active heatsink that is quite big:

[image]

your robot has this heatsink:

[image]

so it seems that you need to remove the back cover in order to install it. Depending on the use-case this could be acceptable or not to keep it open.

cedricgoubard, 2021-07-12:
Hi @maggia80,
Leaving the case open would not be an issue for us. We might try to 3D print a bigger back for the head if needed.
I requested a quote from AdLink Tech for the CMx-SLx.

Assuming the price is not an issue, what would be the next steps?

maggia80, 2021-07-13:
I recommend buying the cable kit as well. 
Once you got it, you should modify the wiring. The robot is not exactly a 2.0, but for the head you can use as a reference this documentation [ iCub 2.0 wiring  ](https://icub-tech-iit.github.io/documentation/icub_wiring/icub2_x/#logic-and-harness-icub-2x-e200).
In the logic schematic you will see the connectors that go to the PC104 ([EMBEDDED LOGIC - PB945+ CPU board - Manual.pdf](https://github.com/robotology/icub-tech-support/files/6806300/EMBEDDED.LOGIC.-.PB945%2B.CPU.board.-.Manual.pdf)), and in the harness, you can see the wires attribute, colors, and position in each connector. It should be enough to modify the wiring for hosting the new board. 
However, before doing that, I strongly recommend you to do the test the USB pen with this board. You probably need to change some of the configurations based on the new HW ( I suppose ethernet cards, and serial for the IMU)

cedricgoubard, 2021-07-13:
Hi @maggia80,
Thank you for the explanation.
What OS should I put on the USB pen for the new card? Should I just reuse the current version (Debian 4.0), or upgrade it to a new one (e.g. [this one](https://github.com/robotology/community/discussions/369) which was recommended some time ago in the discussion)? Or maybe use an even more recent one?

maggia80, 2021-07-14:
Hi @cedricgoubard you can consider to use the Live or to add a SATA HD and go for a standard distribution like we are doing for the new COM-Express board (https://icub-tech-iit.github.io/documentation/icub_operating_systems/icubos/icubos/)

mbrunettini, 2021-07-14:
@cedricgoubard : please not that both iCubLive (the "live" version of iCub head Operative System, used to boot from USB on diskless PC104 board) and iCubOS (the "traditional" version, used to boot from an HD on the new COM-Express board) have not been tested to run on the board you mentioned.
In my humble opinion, you have more chances using the iCubOS which requires an 32GB (or bigger) HD.
See this [link](url)

cedricgoubard, 2021-07-15:
Hi @maggia80, @mbrunettini, thank you for your answers!
I'll let you know how it goes once I hear back from AdLinkTech about the board.

cedricgoubard, 2021-09-13:
Hello, 

I forgot to update this: we heard back about the board. It would take 22 weeks to get it, and since we managed to [update our old board to the latest OS](https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/icub-live-32bits/), we decided not to order it for now.

closed 2021-09-13 by cedricgoubard