robotology/mesh-support #804: iCubDarmstadt01 turning off in the moment I switch on motors

TheCamusean, 2019-04-05:
### Description of the failure

Hello again @julijenv , this has been fast.

Right now, we are facing a strange hardware problem. The steps we are following are these:

1. Turn on the power supplies. Before turning on pc and motor switches they look like the photo I shared you via mail.
2. I turn on the switch of the pc and everything works. I can even ping the pc104.
3. Turn on the switch of the motors. Then, something goes wrong and power suppliers stops proving voltage.

How should we procedure?

Best wishes

### Detailed conditions and logs of the failure

julijenv, 2019-04-05:
Hi @TheCamusean ,
but by your quotation
> Then something goes wrong

What do u mean? Did u launch the robot?
From the picture you sent me the second power supply went in protection. So there was a short circuit somewhere.
Please find me on skype.

TheCamusean, 2019-04-08:
Our iCub's head looks as follows:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

TheCamusean, 2019-04-08:
Better view from sides: 
[image]

[image]

julijenv, 2019-04-12:
hi @TheCamusean ,
did you succeed to remount everything and is everything working again?!??!
let me know

TheCamusean, 2019-04-12:
Hello @julijenv,

Everything smooth and fine.

Thanks alot :dancer:

closed 2019-04-12 by julijenv