robotology/mesh-support #805: Repairing a Force/Torque sensor that has a hardware problem

gsaponaro, 2019-04-09:
### Description of the failure

Hi @marcoaccame, following up from https://github.com/robotology/icub-support/issues/749, we have another related but separate question:

We would like to "repair" the F/T sensor with SN094 that had a hardware problem (on channels 4 and 5), which we extracted from the left arm of our robot, and then we would like to place it into the left leg (that we rarely use).
Is it feasible?

So far, we re-soldered the wires. Unfortunately, the STRAIN board does not show up in the CAN network when we connected it to the leg. We double-checked and verified the continuities involved, and tried several times. Below is a video showing the lights that we observe. After the lights shown in the video, they switch off (no more lights).

Any pointers would be greatly appreciated.

### Detailed conditions and logs of the failure

Video showing the sequence of blinking lights:
https://youtu.be/x9lyuaDmWjM

marcoaccame, 2019-04-09:
Hi @gsaponaro, the blinking is correct and shows that the board bootstraps, waits for 5 seconds (the five slow blinks) and then jumps into application.

The board was running its application when in the arm, thus I don't see reasons why it does not run now.

You should try to run the FirmwareUpdater and perform a discovery on the target bus (the bus #3 as I read in the .xml file of the right leg), but try on other buses as well.

Can you find any strain w/ SN094?

gsaponaro, 2019-04-11:
Thanks for the clarifications here and via email.

Because 1) we cannot find any STRAIN board with CAN-bus, 2) even if we managed to get the 6 ADC readings, we would still have the problem of re-calibrating the sensor, which is a complicated procedure that can only be done by production... we determined that it's easier to forget about this and just purchase a new F/T sensor.

I will contact Fabrizio Larosa to get a quotation.

closed 2019-04-11 by gsaponaro