robotology/mesh-support #809: running iCub_SIM from windows binaries

pfdominey, 2019-04-10:
Greetings,

I have installed the yarp binaries for windows from http://www.yarp.it/download.html

yarp_3.1.0_v14_x86_1
I run the installer, I can run yarpserver, but I do not find where to install the iCub_SIM, and so iCub_SIM is not recognized as a command at the windows cmd prompt.

Probably I just need to download the binary for the simulator, but I dont know where to find it.

thanks
cheers
Peter

traversaro, 2019-04-10:
Hi @pfdominey , 
   the `iCub_SIM` binary is installed by the iCub installer, as documented in http://wiki.icub.org/wiki/Windows:_installation_from_binaries .

pfdominey, 2019-04-10:
Hi
I got the binary for yarp, but I do not see binarys for iCub (only sources) on this page
http://wiki.icub.org/wiki/Downloads
Probably I just cant see it, but could you point me to the link to the binary for window 64 for iCub that has the simulator? thanks!

traversaro, 2019-04-10:
See the second column of the table in http://wiki.icub.org/wiki/Downloads#Windows . In particular, if you installed the 32-bit binary installer for YARP, you should install the 32-bit binary installer for iCub as well, so https://github.com/robotology/icub-main/releases/download/v1.12.0/iCub_1.12.0_v14_x86_1.exe .