robotology/mesh-support #816: Loose mechanical coupling of right arm shoulder joints

Yeshasvitvs, 2019-04-24:
### Description of the failure

We noticed that the right shoulder joints mechanical coupling is loose. We had a lot of broken cables on this shoulder previously and this loose mechanical coupling could be a possible reason.

[image]

CC @DanielePucci @kouroshD @lrapetti

julijenv, 2019-04-29:
Hi @Yeshasvitvs ,
I checked the tension of all cables and effectively some of the inside part where looses. I retightened everything I could posibility find. Let me kow if it is better ths way.

julijenv, 2019-04-30:
any news? @Yeshasvitvs ?!?

Yeshasvitvs, 2019-05-03:
During the pHRI experiments yesterday we did not notice any problems. I can be more conclusive after doing retargeting experiments.

GiulioRomualdi, 2019-05-06:
Today we noticed that the problem is still there. Furthermore, as @julijenv noticed last week, this is caused by a joint of the torso.

Yeshasvitvs, 2019-07-01:
As discussed with @julijenv this issue was not observed in the last test we did for pHRI.
@julijenv will make sure to double check the tension in the cables when mounting the robot parts back. So closing this issue.

closed 2019-07-01 by Yeshasvitvs