robotology/mesh-support #820: Broken  shoulder iCubGenova03

gonzalezJohnas, 2019-05-04:
### Description of the failure
 cable 3,4,6 of the left shoulder broke.

### Detailed conditions and logs of the failure
While performing yoga demo

julijenv, 2019-05-05:
I think it was the right shoulder not the left, Am I right!?! @gonzalezJohnas 
It has been taken from the warehouse
IITCODE Qta Desciption
4760.1 x 1 : shoulder elevation cable N3
4761.1 x 2: shoulder elevation cable N4
4763.1 x 1: shoulder elevation cable N6

gonzalezJohnas, 2019-05-06:
Yes sorry I updated the initial issue, we got the cables this morning the robot is fixed thank you Julien !

julijenv, 2019-05-14:
hi @gonzalezJohnas ,
Could you close the issue if you suceed to repair the left shoulder?
thx in advance

closed 2019-05-14 by julijenv