robotology/mesh-support #821: Black lines in camera feed

fbottarel, 2019-05-14:
### Description of the failure
Black lines started showing up at the top of the image streams

### Detailed conditions and logs of the failure
The Dragonfly cameras output this kind of video stream at 320x240 (image is magnified for better identification of the issue). The top rows of pixels seem corrupted. 

[image]

By resetting the camera via the frameGrabberGui, the image goes back to default parameters (640x480 resolution) and the feed corruption seems no longer present. 

[image]

Rebooting either the cameras, the pc104 or the whole robot does not solve the problem.

julijenv, 2019-05-14:
Hi @fbottarel ,
did you try to run the same test on both cameras?!?!
if yes, i guess it is not hardware related but mainly software - firmware related, if not please do so and if it is just present in one camera, maybe the flat ribbon cable is damaged.
Always in the case of your answer is YES, please try to launch the same test on green iCub, does it make the same thing? I would recommend you to talk to @barbalberto or @Nicogene about that.

Nicogene, 2019-05-14:
Probably the black line is present also in the `640x480` image but it is too small to be seen since the resolution is higher.

It seems not a resolution-dependent problem, `dragonfly2` device is pretty old and nobody changed since a while. The only thing changed are how the images are handled in `yarp`(move semantics etc).

Unfortunately in these day I am at Erzelli. A check good check as @julijenv suggested is to visualize the image 320x240 on the green iCub.

fbottarel, 2019-05-14:
@julijenv both cameras exhibit the same behaviour. I will test this on the green icub as soon as they are done with their tests today.

@Nicogene both resolutions feature the corruption. The corruption disappears if press the "Reset Camera"  button in the FrameGrabberGui, and the camera goes back to a 640x480 resolution (but with no bayer encoding?!)

Also: I dumped some frames with `yarpdatadumper` and the corrupted pixels are still there -> `yarpview` is not the problem.

fbottarel, 2019-05-15:
Update: `camCalib` seems to be the culprit here. Here you can see images streamed from the raw camera feed and from `camCalib`. The top left of the images streamed from `camCalib` seems to be corrupted. 
[image]
@Nicogene halp pls

Nicogene, 2019-05-15:
So i think we can say that it is a software problem then I will move the issue in `icub-main`