robotology/mesh-support #822: Left hand fingertip skin unresponsive

fbottarel, 2019-05-14:
### Description of the failure
Index fingertip skin on the left hand is not working. SkinGui shows no contact. Moreover, the fingernail PCB of the index finger is bent out of shape and one of the wires has detached. 

[image]

### Detailed conditions and logs of the failure

julijenv, 2019-05-15:
Hi @fbottarel ,
I sent a mail to Fabrizio to send someone to resolder this cable.
cheers

julijenv, 2019-05-15:
Hi @fbottarel ,
I hope you do know that I am away from genova until the 26th of June, So please go directly in my lab to get someone to take of that little problem which require a physical interaction.
thx in advnace

fbottarel, 2019-05-15:
Ok, tnx @julijenv!

julijenv, 2019-05-16:
@fbottarel , from my colleague I heard they resoldered the wire, could you please if it works and give feedback to them in the production area, thx in advance

fbottarel, 2019-05-16:
It works as intended, thank you guys as always!

closed 2019-05-16 by fbottarel