robotology/mesh-support #823: Need help to repair broken arm

InriaBrice, 2019-05-14:
### Description of the failure

The cables in the left hand are broken, we want to replace it.
The bearings in the left shoulder are making noise, we want to replace it as well.

### Detailed conditions and logs of the failure

The motors are trying to close the fingers infinitely when we start the robot and he is calibrating himself.

Could we do a skype or something to operate the robot ?
@serena-ivaldi @julijenv

pencoj1, 2019-05-14:
What do you mean by the broken cables? there are literally broken or they simply got disconnected? I guess they simply got disconnected. 

if you open the cover of the tricep, you can see the board mc4. There you can check if there are some disconnected wires. Be sure to have both the motors and the pc104 off when you do some operations there.

For the change fo the bearings we can refer to this issue https://github.com/robotology/icub-support/issues/757

julijenv, 2019-05-15:
Hi @InriaBrice , @InriaBrice ,
if the motor spin it means you dont need to open the lower part of the upper arm.
You jsut need to replace the tendon of the finger broken - it takes 30 min max.
I will be available on skype from 11am to 2pm, everyday, fetch me on skype.
cheers,

InriaBrice, 2019-05-15:
@pencoj1 No no broken cables, for the fingers. There is a motor trying to pull constantly on the index.
@julijenv ok thanks, but for the bearings in the shoulder we have to open something no ?

pencoj1, 2019-06-11:
I guess the problem is just a connection one. The left index and the left middle finger go in HW fault during calibration.  
[Left_Fingers.log](https://github.com/robotology/icub-support/files/3275630/Left_Fingers.log)

In case what are the cables I should check?

julijenv, 2019-06-12:
Hi @pencoj1 ,
coud you try to contact me in between 10am to 12, afterwards I will be in holidays for 1 week and a half.
cheers cu soon

julijenv, 2019-06-12:
CIao @pencoj,
Firs thing first, did the robot fall on its left arm? did a shock occur on the left hand?
If yes and if you are lucky, most probably one of the little pcb attach to the MAIS board above the knuckles of he left hand got misaligned could you double check this? You could check that by looking int othe yarpmotorgui and see if yuo have some values for those 2 joints, and if you move the distal joint for instance if the values change.
(open the dorso cover to inspect that everylayer of boards are parallel)
If yes and if you are unlucky you may have broken some wires, but hopefully not?
If no, just check the little pcb above the knuckles of the left hand all the same. ;)
best of luck

pencoj1, 2019-06-12:
Fixed connection 4M0 and 2M0 as you told me! Great! Now it works. For the change of bearings on the shoulder it would be nice if you can watch as I do. No rush for that. We could have a skype one day that you are available after your holidays. Thanks!    You can close the issue

closed 2019-10-02 by maggia80