robotology/mesh-support #824: Unmounting FT sensors legs and ankles

pencoj1, 2019-05-16:
How to remove the 4 FT sensors on the legs of the iCub?

pencoj1, 2019-05-22:
I was changing the FT sensors. The old one has a connector with 4 wires while the new one 6. How should I proceed?

[image]

julijenv, 2019-05-24:
Hi @pencoj1 ,
sorry for the delay your mail didnt go through outlook (the overall ones from wednesday almost). You have to proceed this way. Either Create an extension from 6 pin female header to connect to the new FT(2) socket and to the other side of the extension you make a 4 pin female header to connect to where was prevously attached your FT(1). The new FT as 2 more pin used to programm the internal chip, you dont need to care about those 2 external pin (purple and yellow) Just connect the 4 internal one (red, green, white and black).
Hope it will help you.
I will be on skype from 11:30am to 1pm/2pm your time (Im in Japan right now, which means +7 hours)
C U if ever needed

pencoj1, 2019-06-03:
I changed the FT sensors of the legs and the feet as you explained me.
I get this error when starting the robot with yarprobotinterface

[LEGS_FT.log](https://github.com/robotology/icub-support/files/3247671/LEGS_FT.log)

Do I need to do smth with the firmwareupdater?

julijenv, 2019-06-04:
Hi @pencoj1 ,
when you got the FT sensors, was it written where which one of them should be attached to on the robot? if not probably you need to change either the can add of the sensor. Please control with the FirmwareUpdater that the can add of every FT board is 13 and attached to can2.
Next please go into your repository and change in iCubNancy01/hardware/FT/ for all the files of the left and right leg. chenge the name of the strain from eboard_strain to strain2 and change the major distri from 1 to 2.
here is the example taken for left_leg - what ou should have initially
[image]

and now what it should look like.
[image]
try to change those 3 settings on the repository you launch the robot ( where you would tell me, just stop the motor (turn off the switch motor) and launch yarprobotinterface from command line on a new shell, and as soon as you see the tons of line printeing stop it with control +C for 2 or 3 times. check the stsrting of the lines and find where the yarprobotinterface grad its setting files.
hope this will help.

pencoj1, 2019-06-04:
Ok fixed for the left_leg-eb6_j03 and right_leg-eb8-j03. 

Now I have the error only on these boards:  

missing eobrd_strain2 BOARD 10.0.1.9:CAN2:1
 from BOARD 10.0.1.9 (right_leg-eb9-j4_5)

missing eobrd_strain2 BOARD 10.0.1.7:CAN2:1
 from BOARD 10.0.1.9 (left_leg-eb7-j4_5)

Sorry I hadn't specified that I had the error also for these boards, before.

I guess their xml file has to be different cause I tried with the same modifications of the ones that now work, but I still have the error.

julijenv, 2019-06-05:
Hi @pencoj1 ,
from what I can see the yarprobotinterface is looking for iD 1 and from what I recall your can ad id 13 so I guess that if you cnahed them to 1 it should be ok (those of the feet) on board 10.0.1.9 and 10.0.1.7, please change their can add with FirmwareUpdater.
let me know

pencoj1, 2019-06-05:
you're right like always!  the can add id is now 13 and I have to change it to 1. I dunno how to change this address on the frimwareupdater tho. Shall I click on <Upload Application> after selecting the strain2 on board 10.0.0.1.9? because is the only button that I can click now together with <change Info> that doesn't do anything useful

julijenv, 2019-06-06:
Hi @pencoj1 , ah oops I totally forgot, to do that you need to enter in super control mode, i do believe you need to add "-- admin" or "-- calib" to be able to see the window that allows you to change the ID.
Hope it will solve the problem.
cheers buddy

fjandrad, 2019-06-06:
--admin is the right option

pencoj1, 2019-06-11:
Now it works! Thx!
I Think I have to do some calibration tho. I tested the sensors with the legs straight (hip roll at 0) and the iCub on the pole with the feet floating in the air.
In the following yarpscopes the legend is (Force. X:red Y:green Z:blue. Torque. X:white Y:yellow  Z:black)
The Left Foot sensor is fine.
The Z Force on the Right Leg seems to be too much.
Right Foot and Left Leg seem to be wrong

[image]
[image]
[image]
[image]

fjandrad, 2019-06-11:
A common procedure for this kind of sensors is zeroing before starting to be used for every task. In the lab instead of a full zero value we estimate the "correct" value for a configuration of the robot with only one contact and remove the offset. Is part of the functionalities of wholebodydynamics. But I do not know if you'd use it or not.

closed 2019-06-12 by pencoj1