robotology/mesh-support #825: Left pinky fingers not moving

xEnVrE, 2019-05-16:
### Description of the failure
During some grasping experiments, in preparation for tomorrow's demo for `Argos Company`, the pinky left fingers stopped moving suddenly.

Encoders seem to be working but we are unable to move the fingers from the `yarpmotorgui`.

We kindly ask to repair the fingers before tomorrow afternoon due to an **important demo** starting at 15.00.

cc @fbottarel

julijenv, 2019-05-17:
Hi @xEnVrE and @fbottarel ,
Could you please check that the cable is not broken?!?! (please spin the motor of pinky and ring finger motor shape Counter ClockWise as from this picture until both fingers close together)
[image]
If they dont close or just one , keep spinning until to got some tension. After getting the tension if just one finger close it means that the 2nd one is broken.
If you can have them closing together, it oucld possibly mean that you need to restart in csome configuration like come back to just before (by rewinding the finger) in order to have them close to the value encoder 0 but not minus something ( do you get values going from 0 to +200 something when visualizing the yarpmotorgui for the pinky finger?).
If it is not enough to get the finger to work after powering OFF and ON the motor and relaunching the yarprobotinterface, it may come from the connector of the motor inside the upper arm. You may need to ask inside the produciton lab that someone take a look at the motor 4M3, check with the jogger board and recrimp if the motor dont close the fingers.
let me know

vtikha, 2019-05-17:
Hi @julijenv , I tried that this morning and only one finger closes. The ring finger is the one that appears to be broken.

julijenv, 2019-05-17:
Hi @vtikha ,
So I guess that if you take the cable running along the finger you would find that it is broken, right? try to pull it out to confirm?
If the cable is broken, you need to find someone able to repair it.
I'm still available to support someone willing to take this challenge.

vtikha, 2019-05-19:
Hi @julijenv , sorry for not replying sooner. The cable was loose and blocked while badly twisted around the pulley,  we had to call for backup (we were expecting visitors) to replace both cables for the pinky and the ring finger (pinaffo). Thank you for the support @julijenv

closed 2019-05-19 by vtikha