robotology/mesh-support #827: Fingers not getting calibrated correctly

Yeshasvitvs, 2019-05-27:
### Description of the failure

On both the hands of the robot, the thumb, index and middle fingers are not getting calibrated correctly. They are always stopped in closed position.

@gabrielenava 

### Detailed conditions and logs of the failure

kouroshD, 2019-07-22:
@julijenv  Currently, when we start `yarprobotinterface`, `joint7: r_hand_fibger` stays in hardware fault, and `joint 14: r_middle_distal` is in `NOT CONFIGURED`. Can you check it please. Thank you.

kouroshD, 2019-07-22:
@julijenv 
Today, when we were working with robot, at a certain point, the robot's right-hand figures start moving, while we did not send any command to the robot. Later, all the right-hand joint went into `hardware fault`, and we could not put them in `idle`.
This behavior happens two times today.
After the first time, I stopped the `yarprobotinterface`, and later we have trouble starting it.

kouroshD, 2019-08-02:
> @julijenv Currently, when we start `yarprobotinterface`, `joint7: r_hand_fibger` stays in hardware fault, and `joint 14: r_middle_distal` is in `NOT CONFIGURED`. Can you check it please. Thank you.

@julijenv  any update? Thanks.

closed 2019-10-12 by pattacini