robotology/mesh-support #830: While having the an encoder broken the other joints connected to the mais board are not stiff.

fjandrad, 2019-06-12:
### Description of the failure
The robot fell while walking and broke an encoder ( right hip yaw), but assuming that the other joints should be fine we attempted to use the robot both in position control and in torque control. In both cases, the other joints connected to the mais board ( hip roll, hip pitch and knee ) are not able to retain the value set by the controller. Therefore the leg can be moved by hand even when it is supposed to be stiff and keep the same position value.

### Detailed conditions and logs of the failure
There does not appear an specific message in the log. Although @valegagge  and @randaz81 observed something on the 4th joint ( knee ) during their inspection of the broken encoder.

[image]

julijenv, 2019-06-27:
Hi @fjandrad ,
did someone check something on this matter?
MAIS board in the leg?!?!? MAIS are located into the hands, I guess you wanted to talk about another board.

traversaro, 2019-06-27:
I think @fjandrad actually meant `EMS`.

fjandrad, 2019-07-01:
Yeah sorry ems board xD. As far as I know they fixed the  broken encoder but did not check this behavior.

julijenv, 2019-07-01:
hi @fjandrad ,
so I'd like to close some of the overall opened issue on GREENCH, could you let me know, if this one is settled now? if yes please close it.

fjandrad, 2019-07-01:
As far as I know it has not been addressed at all. It happened when an encoder was broken. I do not know how to replicate the conditions to verify is still there.

closed 2019-10-12 by pattacini