robotology/mesh-support #835: Failed right_arm BOARD connection while running yarprobotinterface

GiuliaDAngelo, 2019-07-02:
### Description of the failure
I found the yarprobotinterface not working today after @VenusPasandi experiments. 
The yarprobotinterface fails to run reporting a problem related to the right arm. 

### Detailed conditions and logs of the failure
The problem is related to the right arm BOARD and a missing eobrd_mais, as written on the yarplogger below. 

[image]

Detailed error message: 
[image]

I already tried:
* Rebooting the robot but did not solve the issue.
* Pinging the board server (BOARD 10.0.1.4) successfully. 
* Launched the yarprobotinterface from the GUI and from the terminal.
 
We are now running the yarprobotinterface removing all the robot parts from the configuration file except the head and the face.

VenusPasandi, 2019-07-03:
Just to know, I worked with the robot while using all the joints without any issue. I had hardware faults in the fingers but nothing else!
By the way, this morning @GiuliaDAngelo tested the robot again and it worked fine without any error. I mean, the yarprobotinterface run without error and the right arm moved nicely by using yarpmotorgui. So, I think we can close this issue.

closed 2019-07-03 by VenusPasandi

GiuliaDAngelo, 2019-07-03:
Thanks @VenusPasandi !