robotology/mesh-support #836: iCubErzelli01 neck failure

ddetommaso, 2019-07-05:
### Description of the failure
Using the robot for several days with the same experiment and repeating exactly the same movements, in the last few days the neck does unexpected movement before the completion of a requested trajectory.

### Detailed conditions and logs of the failure
Controller used: iKinGazeCtrl
The issue occurs just after the onset of the movement (looking at a certain 3D point). The neck exhibits a shaking trajectory with unexpected peaks of acceleration before reaching the target position.

julijenv, 2019-07-10:
I changed almost the entire flat cable of the right motor of the neck 0M1.

ddetommaso, 2019-07-22:
Unfortunately the neck has problems again, but this time we have the test for showing the issue.

In the following the logs from yarprobotinterface:

[DEBUG]yarprobotinterface running happily 
[DEBUG]yarprobotinterface running happily 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.1 (left_arm-eb1-j0_3) has been silent for 0.0200841 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.3 (right_arm-eb3-j0_3) has been silent for 0.0201364 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.5 (torso-eb5-j0_2) has been silent for 0.0201476 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.20 (head-eb20-j0_1) has been silent for 0.0201547 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.21 (head-eb21-j2_5) has been silent for 0.0202153 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.22 (face-eb22-j0_1) has been silent for 0.0201941 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.25 (left_arm-eb25-j8_11) has been silent for 0.0202165 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.27 (right_arm-eb27-j4_7) has been silent for 0.0202513 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.28 (right_arm-eb28-j8_11) has been silent for 0.0202601 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.29 (right_arm-eb29-j12_15) has been silent for 0.0203149 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.24 (left_arm-eb24-j4_7) has been silent for 0.0201211 sec (its timeout is 0.02 sec) 
[DEBUG]eth::EthMonitorPresence: BOARD 10.0.1.26 (left_arm-eb26-j12_15) has been silent for 0.0201724 sec (its timeout is 0.02 sec) 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD head-eb20-j0_1 IP 10.0.1.20 and nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[WARNING]EthResource::getRemoteValue() cannot have a reply from BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 1 w/ timeout of 0.2 seconds 
[ERROR]  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb20-j0_1 with IP 10.0.1.20  even after 0.200189 seconds: CANNOT PROCEED ANY FURTHER 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD head-eb20-j0_1 (IP 10.0.1.20)  
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD head-eb21-j2_5 IP 10.0.1.21 and nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[WARNING]EthResource::getRemoteValue() cannot have a reply from BOARD head-eb21-j2_5 with IP 10.0.1.21 at attempt # 1 w/ timeout of 0.2 seconds 
[ERROR]  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb21-j2_5 with IP 10.0.1.21  even after 0.20024 seconds: CANNOT PROCEED ANY FURTHER 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD head-eb21-j2_5 (IP 10.0.1.21)  
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD head-eb21-j2_5 IP 10.0.1.21 and nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[WARNING]EthResource::getRemoteValue() cannot have a reply from BOARD head-eb21-j2_5 with IP 10.0.1.21 at attempt # 1 w/ timeout of 0.2 seconds 
[ERROR]  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb21-j2_5 with IP 10.0.1.21  even after 0.200195 seconds: CANNOT PROCEED ANY FURTHER 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD head-eb21-j2_5 (IP 10.0.1.21)  
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD head-eb20-j0_1 IP 10.0.1.20 and nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[WARNING]EthResource::getRemoteValue() cannot have a reply from BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 1 w/ timeout of 0.2 seconds 
[ERROR]  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb20-j0_1 with IP 10.0.1.20  even after 0.200165 seconds: CANNOT PROCEED ANY FURTHER 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD head-eb20-j0_1 (IP 10.0.1.20)  
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD head-eb20-j0_1 IP 10.0.1.20 and nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[WARNING]EthResource::getRemoteValue() cannot have a reply from BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 1 w/ timeout of 0.2 seconds 
[ERROR]  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb20-j0_1 with IP 10.0.1.20  even after 0.200247 seconds: CANNOT PROCEED ANY FURTHER 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD head-eb20-j0_1 (IP 10.0.1.20)  
[ERROR]hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.24. Expected: 390280, Received: 390992, Missing: 712, Prev Frame TX at 1385998381 us, This Frame TX at 1388137381 us 
[ERROR]hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.27. Expected: 389696, Received: 390409, Missing: 713, Prev Frame TX at 1386002407 us, This Frame TX at 1388144407 us 
[... cut ...]
[ERROR]  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb21-j2_5 with IP 10.0.1.21  even after 0.20019 seconds: CANNOT PROCEED ANY FURTHER 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD head-eb21-j2_5 (IP 10.0.1.21)  
^C[INFO][try 2 of 3] Trying to shut down.
[WARNING]Interrupt # 2 # received. 
[INFO]Interrupt received. Stopping all running threads. 
[DEBUG]head-calibrator killing park of device head-mc_wrapper 
[DEBUG]HeadV3_Calibrator : Quitting parking
 
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
Aborted (core dumped)

julijenv, 2019-07-23:
Hi @ddetommaso ,
I checkout the log file and actually this time it is more related to the ETH connexion in between the board 20 and 21 and by so I mean it is another kind of problem as the one of this issue but well it is stll related to the head though :S.

ddetommaso, 2019-07-24:
Just for adding more details, thanks to @adamlukomski we have [videos](https://www.youtube.com/watch?v=ErXqTuA6PYI) showing the issue.

ddetommaso, 2019-07-26:
We're still testing the robot in different conditions, although the issue occurs more often it's not always present. In the next days we'll post here logs and videos to add more info. @julijenv @maggia80

maggia80, 2019-07-26:
Looking at the log, as @julijenv pointed out it seems a problem related to board 20, that is the first of the chain and it is the one that controls the two motors of the neck. It could be a problem related to the ETH cable from the PC to the board 20 or from 20 to 21, or for some reason the board 20 resets. It could be a power issue: voltage drops because of the cable length or because of the big request of current from the motors and the capacitor in the head is not enough or it is unconnected or faulty). Let's exclude the cable length by removing the cable cord extension of the robot. We could remove also the cover of the head and take a video while moving, in order to check the leds of the board 20.

[image]

adamlukomski, 2019-07-26:
How low can we go with the voltage during normal operation for those boards? Currently the "get avsu 0" on the /icub/head/rpc:i reports voltage dropping down to even 11.7 (from initial 13.4) during movements. Also, is there a chance to get the schematics for the board EB20 to see the layout of the power delivery? What does "get avsu" actually measure?

maggia80, 2019-07-29:
Hi @adamlukomski ,
The voltage input for the board EB20 goes from 9V to 32V, 12V is the nominal one. I guess that "get avsu" is providing the Vin measurement. I can provide you the schematic of the board EB20, however, I think is more interesting for you the wiring diagram of the robot. 
[iCub2.5-E3.3.2_5672_Logic.pdf](https://github.com/robotology/icub-support/files/3440911/iCub2.5-E3.3.2_5672_Logic.pdf)
[mc4-plus_rev2_dsn.pdf](https://github.com/robotology/icub-support/files/3440912/mc4-plus_rev2_dsn.pdf)

adamlukomski, 2019-07-29:
New movie: https://www.youtube.com/watch?v=_ZzGgykpsNE

In the subcritical scenarios the reported voltages were dropping down to even 7.6V, during crashes usually the boards were lost before reporting voltages. From what I remember LM5575 is supposed to be able take it even below 6V, but I suppose the KSZ8893 switch will surrender before the STM32F407 does and we will not see a single error message from the boards. Is there by chance any voltage warning in the firmware or anything like that? 

Logs for the movie:
[01_log_manual.txt](https://github.com/robotology/icub-support/files/3441921/01_log_manual.txt)
[01_voltage.txt](https://github.com/robotology/icub-support/files/3441922/01_voltage.txt)
[01_yarprunlog_29_07_2019_09_59_11.log](https://github.com/robotology/icub-support/files/3441923/01_yarprunlog_29_07_2019_09_59_11.log)
[02_log_manual.txt](https://github.com/robotology/icub-support/files/3441924/02_log_manual.txt)
[02_voltage.txt](https://github.com/robotology/icub-support/files/3441925/02_voltage.txt)
[02_yarprunlog_29_07_2019_10_03_05.log](https://github.com/robotology/icub-support/files/3441926/02_yarprunlog_29_07_2019_10_03_05.log)
[03_log_manual.txt](https://github.com/robotology/icub-support/files/3441927/03_log_manual.txt)
[03_voltage.txt](https://github.com/robotology/icub-support/files/3441928/03_voltage.txt)
[03_yarprunlog_29_07_2019_10_06_29.log](https://github.com/robotology/icub-support/files/3441929/03_yarprunlog_29_07_2019_10_06_29.log)
[04_log_manual.txt](https://github.com/robotology/icub-support/files/3441930/04_log_manual.txt)
[04_voltage.txt](https://github.com/robotology/icub-support/files/3441931/04_voltage.txt)
[04_yarprunlog_29_07_2019_10_09_39.log](https://github.com/robotology/icub-support/files/3441932/04_yarprunlog_29_07_2019_10_09_39.log)
[05_log_manual.txt](https://github.com/robotology/icub-support/files/3441933/05_log_manual.txt)
[05_voltage.txt](https://github.com/robotology/icub-support/files/3441934/05_voltage.txt)
[05_yarprunlog_29_07_2019_10_12_31.log](https://github.com/robotology/icub-support/files/3441935/05_yarprunlog_29_07_2019_10_12_31.log)
[06_log_manual.txt](https://github.com/robotology/icub-support/files/3441936/06_log_manual.txt)
[06_voltage.txt](https://github.com/robotology/icub-support/files/3441937/06_voltage.txt)
[06_yarprunlog_29_07_2019_10_17_01.log](https://github.com/robotology/icub-support/files/3441938/06_yarprunlog_29_07_2019_10_17_01.log)
[07_log_manual.txt](https://github.com/robotology/icub-support/files/3441939/07_log_manual.txt)
[07_voltage.txt](https://github.com/robotology/icub-support/files/3441940/07_voltage.txt)
[07_yarprunlog_29_07_2019_10_20_04.log](https://github.com/robotology/icub-support/files/3441941/07_yarprunlog_29_07_2019_10_20_04.log)
[08_log_manual.txt](https://github.com/robotology/icub-support/files/3441942/08_log_manual.txt)
[08_voltage.txt](https://github.com/robotology/icub-support/files/3441943/08_voltage.txt)
[09_log_manual.txt](https://github.com/robotology/icub-support/files/3441944/09_log_manual.txt)
[09_voltage.txt](https://github.com/robotology/icub-support/files/3441945/09_voltage.txt)
[09_yarprunlog_29_07_2019_10_25_13.log](https://github.com/robotology/icub-support/files/3441946/09_yarprunlog_29_07_2019_10_25_13.log)
[10_log_manual.txt](https://github.com/robotology/icub-support/files/3441947/10_log_manual.txt)
[10_voltage.txt](https://github.com/robotology/icub-support/files/3441948/10_voltage.txt)
[10_yarprunlog_29_07_2019_10_28_45.log](https://github.com/robotology/icub-support/files/3441949/10_yarprunlog_29_07_2019_10_28_45.log)
[11_log_manual.txt](https://github.com/robotology/icub-support/files/3441950/11_log_manual.txt)
[11_voltage.txt](https://github.com/robotology/icub-support/files/3441951/11_voltage.txt)
[12_log_manual.txt](https://github.com/robotology/icub-support/files/3441952/12_log_manual.txt)
[12_voltage.txt](https://github.com/robotology/icub-support/files/3441953/12_voltage.txt)
[12_yarprunlog_29_07_2019_10_32_53.log](https://github.com/robotology/icub-support/files/3441954/12_yarprunlog_29_07_2019_10_32_53.log)
[13_log_manual.txt](https://github.com/robotology/icub-support/files/3441955/13_log_manual.txt)
[13_voltage.txt](https://github.com/robotology/icub-support/files/3441956/13_voltage.txt)
[13_yarprunlog_29_07_2019_10_35_22.log](https://github.com/robotology/icub-support/files/3441957/13_yarprunlog_29_07_2019_10_35_22.log)

adamlukomski, 2019-07-29:
Movements used to show the problems: (pure sockets in python3, no python yarp bindings needed)
[enduranceTest.zip](https://github.com/robotology/icub-support/files/3442480/enduranceTest.zip)

adamlukomski, 2019-07-30:
Maybe another question, linked to our extensive list of neck issues - sometimes (quite rare, once a day at most) yaw motor is being randomly disabled due to "overheating" and "temperature protection" (I'll add a log when it again occurs) - how is the temperature of the motors estimated on the boards? I checked and during most movements we are far away from the current limits of 2.0A set for it.

Also, about the voltage drops: the only other voltage measurements available are the ones from the fingers, but they do not change if we start moving the head, so I assume either the issue is only with the head boards or the fingers have their own power supply (what is BDP BP1?). Can we get voltage info from other boards, like 2FOC/EMS?

@maggia80 is there any repository from which I can download schematics for other boards?

julijenv, 2019-07-30:
Hi @adamlukomski ,
From the log files, the experiments adn the confront with the other robot. I can with almost certainty say that the problem is related to crimp or connector entering in the board 10.0.1.20 either the power line or the ethernet cable. I will check that ASA I will be in Erzelli - supposedly this week.

ddetommaso, 2019-07-31:
Hi @julijenv ,

thank you for the feedback. Please, together with @maggia80 let us know if in the meanwhile further investigations are needed from us in order to have a better understanding of the issue and to ensure that any other components are causing the voltage dropping.

julijenv, 2019-08-08:
Just to end up with this issue. The power line was affected and by swapping the 2 power cable from the board 10.0.1.21 and 23 and double check some of the wiring. The stuation got perfect as usual.

closed 2019-08-08 by julijenv