robotology/mesh-support #838: improvement LEDs - iCub's face

amaroyo, 2019-07-08:
Ciao @valegagge, @julijenv, 

As we already discussed, I had a quick chat with @reafrancesco and @ddetommaso about possible improvements in the LEDs' firmware. We came up with the following proposals - it would be nice to have a way to control the following: 

1) To be able to switch on/off all of the LEDs.
2) To be able to control individually each one of the LEDs including color and intensity - this way we can also create new facial expressions. 

In addition, we have the following question: we would like to show very fast changes in the facial expressions - how fast can we control the changes of the LEDs in the board? For example, can we show changes at least at 30Hz?

Thank you a lot!!
Alex.

alessiav, 2019-07-16:
Hi @valegagge, @julijenv 

I am also interested in controlling the color of the LEDs as I have to continue an experiment with subjects and I started with red LEDs (a change of color can influence their perception of the robot). 
In particular, if it is possible to change the color from blue to red tomorrow would be perfect as the day after I have an experiment.. Otherwise when could it be possible to change it?

Thanks a lot
Alessia

valegagge, 2019-07-16:
Hi @alessiav ,
changing of color needs some effort and new developments, even if it seems a very simple change.
Therefore, I cannot give you the new functionality within tomorrow.

All your required improvements will be take in account and will insert in our activity planning. 

Cheers,
Valentina

julijenv, 2019-08-01:
#775

amaroyo, 2019-10-03:
Any updates regarding this? Maybe timing?
Thanks, 
Alex.

pattacini, 2019-10-25:
Hi @amaroyo 

Let me close this issue as it is somewhat a duplicate of #775, as @julijenv pointed it out.
Further discussion will be taking place therein.

closed 2019-10-25 by pattacini