robotology/mesh-support #839: Cannot run yarprobotinterface on iCubGenova04

VenusPasandi, 2019-07-09:
### Description of the failure
Today, I and @aikolina tried to run `yarprobotinterface` on the green but we did not succeed because we got a different type of error almost every time.

Most of the times, we had problems with the `imu` board. Then, we commented `imu` in `icub-wbd.xml` config file. So, we had problems in almost all the boards. Here we attached the picture of error in the `imu` and the errors in other boards.

At the end, we did not succeed to run the robot!

### Detailed conditions and logs of the failure
[image]

[image]

julijenv, 2019-07-10:
Hi @VenusPasandi ,
first of all as it is one of your first issue that you open, I would like to emphasize on the need of the attachment of the entire logfile when you pubilsh an issue, it is at utmost importance to do not have just the ERROR screenshot but also the warning and other details of what make the ERROR messages to pop-up.
best regards, i will be soon next to the robot.

VenusPasandi, 2019-07-10:
Today, I and @julijenv run the `yarprobotinterface` and we didn't have any problem which is related to `imu` or other control boards (except fingers!) and the robot worked nice. So, I close this issue and if it happens again we will reopen it.

traversaro, 2019-07-25:
@VenusPasandi The issue is still open, there is any reason of simply a leftover?

nunoguedelha, 2019-09-11:
Reopening this issue. Just a quick recap of the issue origin:
- https://github.com/robotology/yarp/issues/1994 was initially occuring at the first yarprobotinterface startup after rebooting the head. This issue resulted in NaN measurements without throwing proper errors.
- that issue was fixed by https://github.com/robotology/yarp/pull/2034, which would run 10 tries for reading the IMU data, after what, upon failure, would through an error, actually observed in @VenusPasandi initial description [here](https://github.com/robotology/icub-support/issues/839#issue-465716533):
[image]

Maybe the number of trials is not enough on iCubGenova04.

@Nicogene @barbalberto any ideas?

CC @GiulioRomualdi

Nicogene, 2019-09-17:
I experienced this issue both on `iCubGenova02` and `iCubErzelli02`, but only when running `yarprobotinterface` just after the power on of the boards.
I have no idea what it is due to, we should investigate it, maybe there is some electrical/hw problem :confused:

nunoguedelha, 2019-09-17:
A hardware/electrical issue that would never occur at the second run of the `yarprobotinterface`? hum, it seems quite doubtful to me.
I think whatever time you wait after powering up the CPU and motor boards, the first `yarprobotinterface` run always fails.

closed 2019-10-01 by julijenv