robotology/mesh-support #840: LEDs improper behaviour

amaroyo, 2019-07-10:
Ciao @valegagge, 
cc: @julijenv, @reafrancesco 

I have been testing the new LEDs, and we have some delayed/wrong behavior. For example, using the rpc connection to iCub's emotions to send "set all hap" does not work from the first time always - not all the boards (left_eyebrow, right_eyebrow, mouth) change accordingly. I should send the same command 2, 3, 4 times to make all of the boards change to the new expression. I couldn't find a pattern, sometimes it works even form the first time. The same happens with other facial expressions as well. 

Here in the following link you can find a video of the cycle.sh performed with the new LEDs. You can see that sometimes the facial expressions are wrong (e.g., right_eyebrow does not always change). As a test, I ran the cycle on the robot iCubErzelli03 (the one with the head only and the same LEDs), and it works without any problems. 
[video_cycle](https://www.dropbox.com/sh/mzkzqu63018kysi/AAAGoLhtgU8OZ48hbx-em-Qja?dl=0)

I think there should be some kind of delay, because we manage to have expressions such this one (even though is supercool and I vouch to implement it :smirk: ): 
[image]

Another test I did with Reddy and the iCubErzelli03 is checking the speed of chage using the raw. 

```
#!/bin/bash
  
listM="M01 M02"
listL="L01 L04"
listR="R01 R04"

while true; do
        for a in $listR; do
                echo "set raw $a"
                echo "$a"  | yarp write ... /icub/face/raw/in
                sleep 0.01
        done
done
```

And the results are that iCubErzelli03 performs very well, and Reddy has some problems. You can check the videos here: [videos testing](https://www.dropbox.com/sh/mzkzqu63018kysi/AAAGoLhtgU8OZ48hbx-em-Qja?dl=0)

As I ran exactly the same modules, and scripts... I suppose the problem is either in the firmware version or the boards themselves. 

Let me know if you need some more info!!

Thanks, 
Alex.

valegagge, 2019-07-12:
Hi @amaroyo ,

it seems somthing related to the hardware or related to the frequency of commands.

In the second video, that bout Erzelli03, what happen? Can you describe me the situation?
Moreover, which commands you send in the third video and which is their frequency?

Thanks,
Vale

valegagge, 2019-07-12:
cc @MrAndrea

amaroyo, 2019-07-14:
Ciao @valegagge, 

The videos iCubGenova03.mp4 and iCubErzelli03.mp4 are the output of the same chunk of code - we ran just the following script: 

```
#!/bin/bash
  
listM="M01 M02"
listL="L01 L04"
listR="R01 R04"

while true; do
        for a in $listM; do
                echo "set raw $a"
                echo "$a"  | yarp write ... /icub/face/raw/in
                sleep 0.01
        done
done
```

As you can see in the videos, the Erzelli one keeps up with the commands, while the Genova one is lagging. This makes me guess that we have either a different firmware running on the iCubs or our boards are faulty. 

The biggest issue is that sending commands to the LED boards is not reliable, and we use quite a lot the facial expressions in our experiments. 

Thanks, 
Alex.

amaroyo, 2019-07-15:
cc @julijenv, once you've completed the two other set of boards, put the cables and so on.... can you please give them to @valegagge, so she can test these issues? Thanks!

valegagge, 2019-07-16:
Hi @amaroyo,
If I'm not wrong, your robot hasn't the newest version of head-pc, so maybe the issue is due to less  performance pc.

For your experiments, is it necessary to send commands every 0.01? Maybe if you send commands with a lower frequency, the LEDs behavior is ok.

Moreover about tests you mentioned , I think the better way to solve this issue is to perform test on your robot directly.

Cheers,
Vale

amaroyo, 2019-07-16:
Ciao @valegagge, 

Sorry, maybe I didn't explain myself properly... This is what's currently not working and we should solve asap: 

I run these two modules on the pc104:
```
yarpdev --verbose --name /icub/face/raw --device serial --subdevice serialport --context faceExpressions --from serialport.ini

emotionInterface --name /icub/face/emotions --context faceExpressions --from emotions_rfe.ini
```

I connect them, and I access the port ```yarp rpc /icub/face/emotions/in```

I ran the following commands (each one after 10seconds, and I did a photo of the result). I also checked the workload of the pc104 and is almost non existent.

```
(0) neutral:

(1) set all hap -> ok

(2) set all ang -> mouth wrong

(3) set all ang -> mouth wrong - 1 led off

(4) set all ang -> ok

(5) set all hap -> right eyebrow wrong

(6) set all hap -> ok

(7) set all ang -> right eyebrow wrong

(8) set all ang -> no change, still wrong

(9) set all ang -> no change, still wrong

(10) set all ang -> no change, still wrong 

(11) set all ang -> ok

(12) set all hap -> right eyebrow wrong

(13) set all hap -> no change, still wrong

(14) set all hap -> no change, still wrong

--here I stopped taking pics

(15) set all hap -> no change, still wrong

(16) set all hap -> ok

``` 

Here you can see the results of the expressions:

[image]

Let me know!

Thanks, 
Alex.

valegagge, 2019-07-16:
Hi @amaroyo ,
ok now it is clear! :)

I can test the expressions on our set up within Thursday afternoon.

I will let you know soon.
Valentina

amaroyo, 2019-07-16:
Thanks a lot @valegagge :)

valegagge, 2019-07-18:
Hi @amaroyo,
I did tests and on my setup the issue didn't happen.
Tomorrow morning I try a different test and write you asap.

valegagge, 2019-07-19:
Hi @amaroyo ,
I tested the last released firmware on my setup running the script cub-main/app/faceExpressions/scripts/cycle.sh and all work fine.

So, I think the problem could be on your setup.
Please contact @julijenv to check your robot.

Cheers, 
Valentina

amaroyo, 2019-07-24:
Hello @valegagge, 

Sorry for my late reply, but I was abroad. 

As far as I remember, in the current boards we don't have the last firmware release... when we were checking them in your office, something went wrong and we didn't put the last firmware version (I really don't remember why). Maybe this could be the reason of the problems. 

I will get @julijenv and we will test it - I will let you know as soon as I can!

Thanks a lot!
Alex.

amaroyo, 2019-07-26:
Hello @valegagge, 

Yesterday with @julijenv, we discovered that the problem was in the firmware - the version of the boards that we had was 1.1.0 and the one that you gave us is 1.1.2. We also upgraded the board to the new version and everything is working properly! 

Thanks a lot!

Cheers, 
Alex.

closed 2019-07-29 by amaroyo