robotology/mesh-support #841: iCub.Moscow.01 (#038) broken cable in left arm elbow

svyatoslavdm, 2019-07-12:
### Description of the failure
iCub.Moscow.01 (#038) broken cable/rope in left arm elbow 

### Detailed conditions and logs of the failure
During our tests we've encountered a broken linkage, a wire in the iCub arm where you can see in fig.1. After our initial investigation, we found where it broke from as you can observe in fig.2 . 

Do you have any suggestions on how we can fix this problem ? Or any solution or tool for attaching the cap on the wire again.

Additionally, a hexagonal screw appeared unscrewed and detached which seems that it loosen from the robot, but without mechanical drawings, we cannot locate its hole on the robot to tighten it again. Is there any documentation or drawing of robot screws ?

[image]
fig.1.
[image]
fig.2.

julijenv, 2019-07-15:
Hi @svyatoslavdm ,
From what I can see, tell me if Im wrong by taking a look at that [document](https://svn.code.sf.net/p/robotcub/code/trunk/iCubHardware/mechanics/doc/tendonsShoulder/ShoulderWiring_English.pdf), the broken cable is the number 10. The main question now, will be, do you have some spare parts to restore it? BTW the document shows you how to rewire every single shoulder cable, keep it somewhere it could be from a great help in the near fututre.
let me know

svyatoslavdm, 2019-07-26:
Hi @julijenv ,

> do you have some spare parts to restore it?
We don't:(  So we are wondering: 
1) what kind of spare part is that? is there may be shop where we can buy it?
2) what tool should we use to attach a cap to the wire? is there may be shop where we can buy the tool?

julijenv, 2019-07-26:
Hi @svyatoslavdm ,
actually you have 2 options:
either DIY by buying from our supplier the same cable type and crimps. Later crimp them with a press.
Second option is buying a set of cables. Please send a mail to andrea.derito@iit.it for a quotation.

svyatoslavdm, 2019-08-19:
Hi @julijenv 
Sorry for the very late response.
I will send a mail to Andrea, but could you please provide contacts/links to your supplier as well?

julijenv, 2019-08-20:
Hi @svyatoslavdm ,
ask Andrea even for those details.
thx

julijenv, 2019-08-28:
Hi @svyatoslavdm , just to let you know I just give to my colleague the spare part cable number 10 - you didnt confirm me but from the picture fig.1 your sent, it is the only possible cable missing.
By rereading the message of the failure I noted that you were mentionning a screw which fell on the ground, could you please do me a favor and take a picture I could point on where that may come from.
I thought the mechanical design was on free access and downloadable, but I guess you need CAD designer software to read it. the main page talking about it is [here](http://wiki.icub.org/wiki/RobotCub)
let me know

closed 2019-11-03 by stale[bot]