robotology/mesh-support #846: Boards of right arm goes in timeout

Nicogene, 2019-07-22:
### Description of the failure
The boards of the `right_arm` goes on timeout also simply moving it with the `yarpmotorgui`.
Maybe the eth cables have to be crimped.
I heard that it is a common issue of the eth iCubs.

### Detailed conditions and logs of the failure
```
2491	248,212697	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 411s 29m 817u: (code 0x06000002, par16 0x0000 par64 0x3000000000000001) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
2492	248,334305	ERROR	theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 and nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone
2494	248,334471	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb29-j12_15 with IP 10.0.1.29  even after 0.200137 seconds: CANNOT PROCEED ANY FURTHER
2495	248,334509	ERROR	Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)
2496	248,344412	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 411s 156m 102u: (code 0x00000030, par16 0x0001 par64 0x000000000000005c) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
2511	255,331412	ERROR	theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 and nv ID32 = 0x0100030e -> IND = 3, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone
2513	255,331450	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb29-j12_15 with IP 10.0.1.29  even after 0.200107 seconds: CANNOT PROCEED ANY FURTHER
2514	255,331453	ERROR	Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 3 ) for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)
2515	255,422397	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 418s 229m 818u: (code 0x06000002, par16 0x0000 par64 0x3000000000000001) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
2516	255,533818	ERROR	theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 and nv ID32 = 0x0100030b -> IND = 3, TAG = eoprot_tag_mc_joint_status_target
2518	255,533896	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb29-j12_15 with IP 10.0.1.29  even after 0.20013 seconds: CANNOT PROCEED ANY FURTHER
2519	255,533930	ERROR	embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)  joint  3
2520	255,543901	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 418s 356m 102u: (code 0x00000030, par16 0x0002 par64 0x000000000000007d) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
2526	258,018518	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 420s 829m 818u: (code 0x06000002, par16 0x0000 par64 0x3000000000000001) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
2528	258,130293	ERROR	theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 and nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone
2530	258,140407	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb29-j12_15 with IP 10.0.1.29  even after 0.200086 seconds: CANNOT PROCEED ANY FURTHER
2531	258,140521	ERROR	Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)
2532	258,140561	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 420s 956m 102u: (code 0x00000030, par16 0x0001 par64 0x0000000000000080) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
2533	258,538134	ERROR	theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 and nv ID32 = 0x0100020b -> IND = 2, TAG = eoprot_tag_mc_joint_status_target
2535	258,538224	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb29-j12_15 with IP 10.0.1.29  even after 0.200107 seconds: CANNOT PROCEED ANY FURTHER
2536	258,548265	ERROR	embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)  joint  2
2537	258,548338	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 421s 363m 101u: (code 0x00000030, par16 0x0001 par64 0x0000000000000087) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
2538	258,618875	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 421s 429m 818u: (code 0x06000002, par16 0x0000 par64 0x3000000000000001) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
2539	258,759815	ERROR	theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 and nv ID32 = 0x0100030b -> IND = 3, TAG = eoprot_tag_mc_joint_status_target
2541	258,759850	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb29-j12_15 with IP 10.0.1.29  even after 0.200114 seconds: CANNOT PROCEED ANY FURTHER
2542	258,759856	ERROR	embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)  joint  3
2549	261,140632	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 423s 954m 101u: (code 0x00000030, par16 0x0001 par64 0x000000000000008b) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
2556	264,412535	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 427s 229m 818u: (code 0x06000002, par16 0x0000 par64 0x3000000000000001) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
2557	264,544959	ERROR	theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 and nv ID32 = 0x0100000b -> IND = 0, TAG = eoprot_tag_mc_joint_status_target
2559	264,545162	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb29-j12_15 with IP 10.0.1.29  even after 0.200096 seconds: CANNOT PROCEED ANY FURTHER
2560	264,545191	ERROR	embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)  joint  0
2561	264,545228	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 427s 362m 101u: (code 0x00000030, par16 0x0001 par64 0x0000000000000099) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
```

julijenv, 2019-07-22:
Hi @Nicogene ,
I doubt i come from the eth cable because the entire robot will stop as all the board are in daisy chain. Could you run FirmwareUpdater when the robot has this behaviour and check if the board 29 is responding. From the error report you sent me, that the only board showing strange behaviour.
LEt me knwo

Nicogene, 2019-07-22:
Hi @julijenv ,

I stopped `yarprobotinterface` and re-run it and I get other errors during calibration, despite it calibrates fine. Opening the `yarpmotorgui` everything seems ok.

```
2659    112.014440    ERROR     from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 734s 228m 551u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2660    112,014550    ERROR     from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 734s 228m 684u: (code 0x02000010, par16 0x0003 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2661    112,218971    ERROR     from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 734s 430m 709u: (code 0x02000010, par16 0x0003 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2662    112,219025    ERROR     from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 734s 434m 676u: (code 0x02000010, par16 0x0000 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2677    112,863743    ERROR     from BOARD 10.0.1.28 (right_arm-eb28-j8_11), src LOCAL, adr 0, time 735s 88m 44u: (code 0x02000010, par16 0x0002 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2678    112,873887    ERROR     from BOARD 10.0.1.28 (right_arm-eb28-j8_11), src LOCAL, adr 0, time 735s 91m 31u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2679    112,873952    ERROR     from BOARD 10.0.1.28 (right_arm-eb28-j8_11), src LOCAL, adr 0, time 735s 91m 166u: (code 0x02000010, par16 0x0003 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2680    113,077412    ERROR     from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 735s 295m 460u: (code 0x02000010, par16 0x0003 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2681    113,077494    ERROR     from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 735s 296m 424u: (code 0x02000010, par16 0x0000 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2682    113,077515    ERROR     from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 735s 296m 544u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2683    113,077529    ERROR     from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 735s 296m 669u: (code 0x02000010, par16 0x0002 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2684    113,178741    ERROR     from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 735s 392m 688u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
2773    113,872935    ERROR    [ [01;32mDEBUG [00m]left_leg-calibrator starting calibration of device left_leg-mc_wrapper
2777    113,883013    ERROR    Right_Arm_Calibrator : detected an hardware fault on joint  10  before starting the calibration procedure!
2781    113,883099    ERROR    Left_Arm_Calibrator : detected an hardware fault on joint  11  before starting the calibration procedure!
2782    113,883112    ERROR    Right_Arm_Calibrator : detected an hardware fault on joint  12  before starting the calibration procedure!
2783    113,883148    ERROR     [01;32mDEBUG [00m]right_leg-calibrator starting calibration of device right_leg-mc_wrapper
2787    113,883208    ERROR    Right_Arm_Calibrator : detected an hardware fault on joint  14  before starting the calibration procedure! [ [01;31mERROR [00m]Left_Arm_Calibrator : detected an hardware fault on joint  13  before starting the calibration procedure!
2790    113,883255    ERROR    Left_Arm_Calibrator : detected an hardware fault on joint  14  before starting the calibration procedure!
2791    113,883270    ERROR    Left_Arm_Calibrator : detected an hardware fault on joint  15  before starting the calibration procedure!
2796    113,883339    ERROR    Right_Arm_Calibrator : detected an hardware fault on joint  15  before starting the calibration procedure!
2931    114,598227    ERROR     from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 736s 806m 699u: (code 0x02000010, par16 0x0002 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
3027    117,008899    ERROR     from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 739s 229m 548u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
[... cut ...]
3056    118,180795    ERROR     from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 740s 393m 688u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
3073    119,603062    ERROR     from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 741s 807m 699u: (code 0x02000010, par16 0x0002 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
```

Unfortunately it happens quite often, I don't know if it is related to the problem I posted before.

Nicogene, 2019-07-22:
>  Could you run FirmwareUpdater when the robot has this behaviour and check if the board 29 is responding. 

I launched the `FirmwareUpdater` and it says `application(RUNNING)` for the board 10.0.0.29

marcoaccame, 2019-07-22:
Hi all,

what happens is that the ETH link in entrance to the board 10.0.1.29 is affected by electrical noise which produces CRC errors with loss of packets and triggers a lot of errors.

here is some more explanations.

the board BOARD 10.0.1.29 is surely communicating because at time 248,212697 it sends up a diagnostics message:

```
2491	248,212697	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, 
time 411s 29m 817u: (code 0x06000002, par16 0x0000 par64 0x3000000000000001) -> 
ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 
1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 
0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
````

This diagnostics message tells that the board 10.0.1.29, which is a mc4plus called right_arm-eb29-j12_15, has detected a CRC error in UDP transmission w/ consequent lost of one UDP packet.

The lost packet is in the link which goes from board 10.0.1.25 to the board 10.0.1.26.

The loss of the this packet generates a sequence of error messages coming from yarprobotinterface:

the following tells that a request of the value of eoprot_tag_mc_joint_status_core_modes_ismotiondone has timed-out (due to the loss of the above UDP packet).
```
2492	248,334305	ERROR	theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 and nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone
``` 
and from here otehr errors triggered by the layers of software above:

```
2494	248,334471	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb29-j12_15 with IP 10.0.1.29  even after 0.200137 seconds: CANNOT PROCEED ANY FURTHER
2495	248,334509	ERROR	Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 0 ) for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)
```

After a while, when teh board 10.0.1.29 receives a further UDP packet from yarprobotinterfce the board detects that one UDP packets is missing. 
````
2496	248,344412	ERROR	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 411s 156m 102u: (code 0x00000030, par16 0x0001 par64 0x000000000000005c) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
```
etc.

After this first CRC error, there is a second CRC error at time 255,422397 which triggers a lot of further errors.

julijenv, 2019-07-22:
Hi @Nicogene , from I could understand thx to the help to @marcoaccame , is that the network cable junction in between the board 28 and 29 have a weak connection.
probable some crimp need to be redone.
I'll be on the Erzelli on thursday, will ou be there too?

Nicogene, 2019-07-22:
Thanks @marcoaccame for the explanation !

@julijenv yes we will @ Erzelli for the whole day, let us know when you will come. Many thanks!

Nicogene, 2019-08-02:
Today we fixed it with @julijenv, closing

closed 2019-08-02 by Nicogene