robotology/mesh-support #852: iCub left shoulder broken

dariopasquali, 2019-08-06:
Ciao @julijenv the cable num 6 of the left shoulder is broken and we don't have any spare one.
Could you please help us?
Thank you,
Dario

julijenv, 2019-08-07:
Hi @dariopasquali ,
from what I recall I show you how to recable this joint and actually it is the simple one of the all the shoulder cable so if you just can have someone to passs by here in san quirico to get the cable you can have it fxed in no time, the other possible way it to come to pick me up but I guess it would be quite an harass to do this way.
let me know

dariopasquali, 2019-08-07:
Hi @julijenv,
Thank you! I'm coming to San Quirico to take the cable.
Just a thing, I've never been in San Quirico, so can you tell me where I have to go?

Thank you, 
Dario

dariopasquali, 2019-08-08:
I @julijenv ,
I fixed the left shoulder, everything works fine!

Thank you!

closed 2019-08-08 by dariopasquali