robotology/mesh-support #853: Replacement of FT sensor on the left ankle

GiulioRomualdi, 2019-08-07:
In the last days we noticed that the data streamed by the left ankle FT sensor are affected by a constant offset on the x-axis. 
The aferomentioned offset is not present on the right ankle FT.

Discussing with @DanielePucci and @fjandrad we concluded that the problem may be related to the sensor itself.

cc @prashanthr05 @nunoguedelha

julijenv, 2019-08-09:
After having discussed with @maggia80 and @fjandrad and to ease the process for the deadline of the experiment of @GiulioRomualdi, we decided to remount another FT sensor on the left ankle, but the one unmounted seems to be ok except by the offset increasing on x by a factor of 1/10 of the one on the Z, which suggest a problem related to calibration. @fjandrad accepted to do an in board (in situ) recalibration of the sensor (S/N:002) later in order to help us in the debugging (by confronting both matrixes).

GiulioRomualdi, 2019-08-09:
Thanks guys!

fjandrad, 2019-08-10:
We verified if the problem observed on the robot disappeared by replacing the sensor. Unfortunately, the problem did not go away. Therefore the sensor is likely ok and does not require calibration, but we need to check why the sensor is measuring that when mounted on the foot. There is the possibility that for some reason the sensor is not actually parallel to the ground or has some stress that occurs when the robot is standing on the foot.

We did some tests with the IMU on the ft sensor at the feet and there seems to be some inclination this will require a deeper look. Anyway the sensor was replaced so we can probably close this issue.

closed 2019-08-12 by julijenv