robotology/mesh-support #854: Left leg FT streams a constant value after updateTemperature error

GiulioRomualdi, 2019-08-13:
### Description of the failure
The left hip FT starts to stream a constant value.
```
-33.39404296875 17.9738159179688 86.6190795898438 -0.5899658203125 1.4947509765625 -0.29541015625
-33.39404296875 17.9738159179688 86.6190795898438 -0.5899658203125 1.4947509765625 -0.29541015625
-33.39404296875 17.9738159179688 86.6190795898438 -0.5899658203125 1.4947509765625 -0.29541015625
-33.39404296875 17.9738159179688 86.6190795898438 -0.5899658203125 1.4947509765625 -0.29541015625
-33.39404296875 17.9738159179688 86.6190795898438 -0.5899658203125 1.4947509765625 -0.29541015625
-33.39404296875 17.9738159179688 86.6190795898438 -0.5899658203125 1.4947509765625 -0.29541015625
-33.39404296875 17.9738159179688 86.6190795898438 -0.5899658203125 1.4947509765625 -0.29541015625
```
This strange behavior happened four time in the last three day. Checking the error we noticed the following error
```
185842	102182.200814	ERROR	embObjFTsensor: BOARD left_leg-eb6-j0_3 (IP 10.0.1.6)  updateTemperature: I expect 1 item, but I received  2
```
After restarting the motors it seems to be working again.

### Detailed conditions and logs of the failure
Here you can find the complete log:
[log_icub-head_yarprobotinterface_8108.txt](https://github.com/robotology/icub-support/files/3497628/log_icub-head_yarprobotinterface_8108.txt)

cc @marcoaccame @fjandrad

julijenv, 2019-10-01:
equal to #848 - so I closed it

GiulioRomualdi, 2019-10-14:
The problem happens again today. 

This is the logger: [icub_ft_temperature.txt](https://github.com/robotology/icub-tech-support/files/3725158/icub_ft_temperature.txt)

@S-Dafarra @prashanthr05 @marcoaccame

marcoaccame, 2019-10-23:
It will be responsibility to deal w/ this issue of the people of the icub-tech task force under coordination of @julijenv.

In the meantime I will have a close look at the log file.

julijenv, 2019-10-24:
hi @S-Dafarra , @MiladShafiee , @GiulioRomualdi , @prashanthr05 ,
As requested yesterday, to resolve this issue I would like that you perform a simple test.
Please do yoga until the robot got the above mentionned problem and after closing the yarprobotinterface please relaunch it later on WITHOUT rebooting the motor.
WE NEED ABSOLUTELY THIS TEST TO BE PREFORMED TO FIND OUT THE SOURCE OF THE PBLM.
thx in advance
cc @valegagge

valegagge, 2019-10-29:
Hi @S-Dafarra ,
could you tell me firmware versione of strain-2 boards mounted on green iCub, please?

thanks a lot!

PS: @GiulioRomualdi @S-Dafarra As we talked f2f yesterday, in the next days you and your collegues simply run yarprobotinterface on robot configuring all strain2 available services and check if one or more strain2 board stop sending values (temperatures, ft, etc) and in this case attach the log to this issue.

Maybe you already know, but you can run yarprobotinterface and redirect the output on screen and on a file also using `yarprobotinterface 2>&1 | tee logfilename.txt`

cheers 
Valentina

S-Dafarra, 2019-10-29:
> Hi @S-Dafarra ,
could you tell me firmware versione of strain-2 boards mounted on green iCub, please?

@nunoguedelha can you help us on this?

valegagge, 2019-10-31:
I gave a look to the given logs, and I deduced that the problem could be due to a crash of strain board.
I'll be sure after the above test will be performed.

Since we have not a dump core on embedded boards (for obvious reasons), is difficult to catch the bug. I suppose that a bug in the CAN driver can cause the the crash. Therefore I reported the bug in this [issue](https://github.com/robotology/icub-firmware/issues/102) . After fixing it, we could verify if strain still crashes. I think this bug should be fixed anyway. 

In order to understand if such fix resolve this issue, could be useful understand in details the causes and how often it happen. So please, let me know the result of mentioned tests.

GiulioRomualdi, 2019-11-08:
@valegagge we were finally able to collect the loggers:
1. This is the logger related to the crash [log_icub-head_yarprobotinterface_5872_failing.txt](https://github.com/robotology/icub-tech-support/files/3825883/log_icub-head_yarprobotinterface_5872_temperature_failing.txt)

2. Then we tried to restart the robotinterface without rebooting the motors. The interface didn't start. This is the logger: [log_icub-head_yarprobotinterface_6232_after_failing.txt](https://github.com/robotology/icub-tech-support/files/3825884/log_icub-head_yarprobotinterface_6232_after.txt)

@nunoguedelha thanks!

GiulioRomualdi, 2019-11-09:
Checking the loggers in https://github.com/robotology/icub-tech-support/issues/854#issuecomment-551952486 seems that this time the FT on the **right** leg crashed.

valegagge, 2019-11-11:
@GiulioRomualdi and @nunoguedelha 
Thank you for the test.

I read both and now I'm sure the issue is due to a bug in the firmware of strain2. 
In order to catch it, we need to create a setup that can point out it always in a deterministic manner.

S-Dafarra, 2019-11-18:
Related issue: https://github.com/robotology/icub-firmware/issues/102

DanielePucci, 2019-11-18:
Thanks a lot @valegagge for the effort. Unfortunately, addressing this issue is important for demo success since the problem keeps occurring. 

We are in the final testing for the today's RAI demos and the issue keeps occurring making the robot fall. 

CC
@marcoaccame @maggia80

marcoaccame, 2019-12-18:
i close this issue, as i it is closed the robotology/icub-firmware#102 as well.

closed 2019-12-18 by marcoaccame