robotology/mesh-support #855: Right and left arm go in HW fault running the demoRedBall

Nicogene, 2019-08-23:
### Description of the failure
As soon as we run the `redBallDemo` both right and left arm go in HW fault immediately and it is impossible to recover them, unless shutting down and restarting the `yarprobotinterface`.

### Detailed conditions and logs of the failure
```
1812    79,043448    ERROR     from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 360s 927m 594u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1813    79,043488    ERROR     from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 360s 927m 734u: (code 0x02000007, par16 0x0001 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1814    79,043498    ERROR     from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 360s 927m 868u: (code 0x02000007, par16 0x0002 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1815    79,043506    ERROR     from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 360s 928m 637u: (code 0x02000007, par16 0x0003 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1816    79,043512    ERROR     from BOARD 10.0.1.24 (left_arm-eb24-j4_7), src LOCAL, adr 0, time 361s 55m 763u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1831    79,114229    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 360s 997m 620u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1832    79,114239    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 360s 997m 760u: (code 0x02000007, par16 0x0001 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1833    79,114246    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 360s 997m 895u: (code 0x02000007, par16 0x0002 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1834    79,114253    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 360s 998m 656u: (code 0x02000007, par16 0x0003 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1835    79,114259    ERROR     from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src LOCAL, adr 0, time 361s 129m 746u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1839    79,154482    ERROR     from BOARD 10.0.1.24 (left_arm-eb24-j4_7), src LOCAL, adr 0, time 361s 163m 763u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1873    79,316209    ERROR     from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src LOCAL, adr 0, time 361s 330m 746u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1874    79,326372    ERROR     from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 361s 208m 594u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1876    79,326456    ERROR     from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 361s 208m 734u: (code 0x02000007, par16 0x0001 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1877    79,326475    ERROR     from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 361s 208m 868u: (code 0x02000007, par16 0x0002 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1879    79,326508    ERROR     from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 361s 210m 610u: (code 0x02000007, par16 0x0003 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1886    79,457919    ERROR    embObjMotionControl::checkRemoteControlModeStatus(0, 7565168) is unable to check the control mode of BOARD left_arm-eb1-j0_3 IP 10.0.1.1 because it is now in HW_FAULT
1887    79,467951    ERROR    embObjMotionControl::checkRemoteControlModeStatus(1, 7565168) is unable to check the control mode of BOARD left_arm-eb1-j0_3 IP 10.0.1.1 because it is now in HW_FAULT
1888    79,468040    ERROR    embObjMotionControl::checkRemoteControlModeStatus(2, 7565168) is unable to check the control mode of BOARD left_arm-eb1-j0_3 IP 10.0.1.1 because it is now in HW_FAULT
1889    79,468070    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 361s 354m 620u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1890    79,468100    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 361s 354m 760u: (code 0x02000007, par16 0x0001 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1891    79,468127    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 361s 354m 894u: (code 0x02000007, par16 0x0002 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1893    79,478144    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 361s 356m 635u: (code 0x02000007, par16 0x0003 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
1894    79,478178    ERROR    embObjMotionControl::checkRemoteControlModeStatus(3, 7565168) is unable to check the control mode of BOARD left_arm-eb1-j0_3 IP 10.0.1.1 because it is now in HW_FAULT
[... cut ...] ion mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
2036    109,482060    ERROR     from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 391s 362m 640u: (code 0x02000007, par16 0x0003 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .

```

julijenv, 2019-08-23:
Hi @Nicogene , could youp lease attach the whole log file please.
From the ERROR messages it seems that both FT sensor form both arm are not responding, could you check if you can see them with FrimwareUpdater for CAN2 on 10.0.1.1 and 10.0.1.3? Is the ID number the one it should? both shoudl be ID number 13, Did uou update the boards by chance?

Nicogene, 2019-09-04:
Hi @julijenv ,
I checked and the ID is 13. We updated the firmware long time ago (approx 6 months) and we never had problems running the `demoRedBall`.

I tried to read from the ports `/icub/left_arm/analog:o` and `/icub/right_arm/analog:o` and I keep receiving data also after the hardware fault.

Today I did another trial I attach the log of `yarprobotinterface`, `wholebodydynamics` and `gravityCompensator`, the only modules that gave errors.

**wholeBodyDynamics**
```
7822	542,122954	ERROR	inverseDynamics thread lost connection with iCubInterface.
7824	542,133007	ERROR	inverseDynamics thread lost connection with iCubInterface.
7826	542,143086	ERROR	inverseDynamics thread lost connection with iCubInterface.
7829	542,153182	ERROR	inverseDynamics thread lost connection with iCubInterface.
7831	542,163250	ERROR	inverseDynamics thread lost connection with iCubInterface.
7833	542,173326	ERROR	inverseDynamics thread lost connection with iCubInterface.
7835	542,183417	ERROR	inverseDynamics thread lost connection with iCubInterface.
7837	542,193503	ERROR	inverseDynamics thread lost connection with iCubInterface.
```
 **gravityCompensator**
```
3524	217,852574	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3587	219,485478	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3589	219,505800	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3591	219,526026	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3593	219,546249	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3595	219,566512	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3597	219,586905	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3599	219,606982	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3601	219,627182	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3603	219,647402	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3605	219,667704	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3607	219,687969	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3609	219,708260	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3611	219,728518	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3613	219,748871	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3615	219,769126	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3617	219,789333	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3619	219,809544	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3621	219,829802	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3623	219,850148	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3625	219,870433	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3665	221,076519	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3934	224,510633	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
3958	225,303696	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4084	229,292347	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4110	229,838882	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4112	229,859071	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4114	229,879301	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4116	229,899676	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4118	229,920156	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4121	229,940386	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4123	229,961707	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4125	229,981990	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4127	230,002131	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4129	230,022487	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4131	230,042757	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4134	230,063114	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4137	230,083338	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4139	230,103561	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4141	230,123702	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4143	230,143856	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
4144	230,143869	ERROR	gravityCompensator module lost connection with iCubInterface, now closing...

```

**yarprobotinterface**

```
6426	421,820851	ERROR	Virtual analog sensor timeout!! No new value received for more than  0.0810187  secs.
6424	421,458479	ERROR	Virtual analog sensor timeout!! No new value received for more than  0.0800009  secs.
6423	421,316635	ERROR	Virtual analog sensor timeout!! No new value received for more than  0.0810115  secs.
6422	420,951504	ERROR	Virtual analog sensor timeout!! No new value received for more than  0.0809512  secs.
6419	420,780440	ERROR	Virtual analog sensor timeout!! No new value received for more than  0.0809846  secs.
[... cut ...]  interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
6397	419,308453	ERROR	 from BOARD 10.0.1.24 (left_arm-eb24-j4_7), src LOCAL, adr 0, time 447s 763m 587u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
```

Nicogene, 2019-10-02:
After the hints provided by @traversaro(thanks a lot), this morning I run the `demoRedBall` but checking before that WBD and gravityCompensator were actually up and running. It worked smoothly, no more HW fault.

The HW fault is actually a secure guard if `demoRedBall` does not receive info from those to executables, in this case the diagnostic messages was not so clear (cc @marcoaccame, @vtikha)

Closing, thanks @julijenv

closed 2019-10-02 by Nicogene