robotology/mesh-support #856: iCubGenova04: Superbuild update to latest branches

nunoguedelha, 2019-08-26:
### Updating the `superbuild` repos
The commits before the update have been tagged `superbuild_23_08_2019` on every repo. The commit in the `robotology-superbuild` repo has been tagged the same way, but with an additional message including the status (commit SHA, branch name pointing to the HEAD, untracked files list, etc).
Any changes found in the working tree have been stashed. the authors of those changes are kindly invited to unstash them and either discard them or properly commit the changes with proper comments if required.

I'm not sure who did what, so I'm listing here potential authors:
@fjandrad @Yeshasvitvs @kouroshD @GiulioRomualdi 
If you have ongoing changes, please unstash and commit, once the main update is complete. I'll let you know. thanks!

The changes are listed below (status prior being stashed by me)...

### Changes previously in working tree (now stashed) or untracked

#### icub-models
On branch `master`
Untracked files:
```
	iCub/robots/iCubGenova04/model_noLeftFootFT.urdf
	iCub/robots/iCubGenova04/model_nofeet.urdf
	iCub/robots/iCubGenova04/model_nofeet.urdf~
	iCub/robots/iCubGenova04/model_nolegsft.urdf
	iCub/robots/iCubGenova04_plus/#model.urdf#
```

#### walking-controllers
On branch `devel_get_torque`
Changes not staged for commit:
```
	modified:   app/robots/iCubGenova04/dcm_walking/joypad_control/plannerParams.ini
	modified:   app/robots/iCubGenova04/dcm_walking/joypad_control/zmpControllerParams.ini
	modified:   modules/Walking_module/src/TrajectoryGenerator.cpp
```

#### ICUB
On branch `flex_motor_input`
Changes not staged for commit:
```
	modified:   src/libraries/icubmod/dragonfly2/common/DragonflyDeviceDriver2.h
	modified:   src/modules/ctpService/main.cpp
```

#### walking-teleoperation
On branch `master`
Untracked files:
```
	app/robots/iCubGenova04/.oculusConfig.ini.swp
```
I recovered the changes in the swap file, such that they appeared in the working tree as follows.
Changes not staged for commit:
```
	modified:   app/robots/iCubGenova04/oculusConfig.ini
```

#### YARP
On branch `devel`
Changes not staged for commit:
```
	modified:   src/devices/imuBosch_BNO055/imuBosch_BNO055.cpp
```

#### robots-configuration
On branch `devel_iCubGenova04`
Untracked files:
```
	iCubGenova04/#general.xml#
	iCubGenova04/clearWBTyarpPorts
	iCubGenova04/debugrobotint.txt
	iCubGenova04/estimators/floatingbaseestimator.xml
	iCubGenova04/estimators/wholebodydynamics_backup.xml
	iCubGenova04/estimators/wholebodydynamics_footMatrices.xml
	iCubGenova04/estimators/wholebodydynamics_iRonCub-noLegs.xml
	iCubGenova04/estimators/wholebodydynamics_noFeet.xml
	iCubGenova04/estimators/wholebodydynamics_noLeftFootFT.xml
	iCubGenova04/estimators/wholebodydynamics_noLegsFT.xml
	iCubGenova04/estimators/wholebodydynamics_waist_imu_original.xml
	iCubGenova04/hardware/motorControl/right_arm-eb3-j0_3-mc.xml.save
	iCubGenova04/icub_no_arms.xml
	iCubGenova04/icub_no_wbd_noimu.xml
	iCubGenova04/icub_test_yaw.xml
	iCubGenova04/icub_wbd_noLeftFootFT.xml
	iCubGenova04/icub_wbd_nofeetFT.xml
	iCubGenova04/icub_wbd_nolegsFT.xml
	iCubGenova04/icub_wbd_noskin_noxsens_inertials.xml
	iCubGenova04/icub_wbd_preprocessor_log.xml
```
This was the status before I stashed everything.

## The update is ongoing.

nunoguedelha, 2019-08-26:
Cleaned-up stashes on:
- **ICUB**

Re-applied all other stashes except 
- **walking-teleoperation**

`robots-configuration` in branch `fix/secondary-cal-ft-2bcb-cisco` now.

Please review stash on **walking-teleoperation** and all files in working-tree or untracked on the other repos.

Please commit changes to a branch and pull-request it by wednesday 28 August, except eventually for `icub-models`.

Remember that clutered repos slow down the workflow and the repos maintenance!
Not commiting files creates a significant risk of loosing significant fixes and waisting time!

@fjandrad @Yeshasvitvs @kouroshD @GiulioRomualdi

nunoguedelha, 2019-08-29:
We ran into the following issues at the robot interface startup:
- https://github.com/robotology/icub-firmware/issues/93, https://github.com/robotology/icub-firmware-build/issues/10 => fixed by https://github.com/robotology/icub-firmware/pull/94, released in  the 2FOC binary [icub-firmware-build/blob/CAN/2foc/2foc.hex](https://github.com/robotology/icub-firmware-build/blob/46d3f477b566be0b01d34409987e5d259abe36bb/CAN/2foc/2foc.hex).
- https://github.com/robotology/icub-firmware/issues/95. The analysis is ongoing, a workaround allowed to run the robot interface all the way through.

With the workaround for https://github.com/robotology/icub-firmware/issues/95, the **Balancing on Two Feet** and the **Yoga** are working fine.

=> Position and Torque control using output PWM as previously.
=> 2FOC controller running at 40kHz.

Remaining actions are the cleanup of the repos commented [here](https://github.com/robotology/icub-support/issues/856#issuecomment-524878972).

nunoguedelha, 2019-09-10:
https://github.com/robotology/icub-firmware/issues/95 has been moved to https://github.com/robotology/icub-main/issues/597, fixed and tested.

Thanks @valegagge and @fjandrad .

**ICUB** (`devel`) and **walking-teleoperation** (`master`) have been cleaned-up.

#### robots-configuration

The following files have been committed to the branch `workaround/avoid-using-FT-on-feet` and pushed to @GiulioRomualdi fork https://github.com/GiulioRomualdi/robots-configuration.git:
```
iCubGenova04/estimators/wholebodydynamics_noFeet.xml
iCubGenova04/estimators/wholebodydynamics_noLeftFootFT.xml
iCubGenova04/estimators/wholebodydynamics_noLegsFT.xml
iCubGenova04/icub_wbd_noLeftFootFT.xml
iCubGenova04/icub_wbd_nofeetFT.xml
iCubGenova04/icub_wbd_nolegsFT.xml
```
The rest of the changes have been pull requested [here](https://github.com/robotology/robots-configuration/pull/118).

Repo is now on branch `fix/secondary-cal-ft-2bcb-cisco`.
Going back to branch `devel_iCubGenova04` after merge of https://github.com/robotology/robots-configuration/pull/118.

#### walking-controllers

Branch `devel_get_torque` to be PRed and merged by @GiulioRomualdi today. After that we go back to `devel` branch.

#### icub-models

The following files have been committed to the branch `workaround/avoid-using-FT-on-feet` and pushed to @GiulioRomualdi fork https://github.com/GiulioRomualdi/icub-models.git:
```
iCub/robots/iCubGenova04/model_noLeftFootFT.urdf
iCub/robots/iCubGenova04/model_nofeet.urdf
iCub/robots/iCubGenova04/model_nolegsft.urdf
```
Repo is now on branch `master`.

#### YARP

On branch `devel`
Changes not staged for commit:
```
	modified:   src/devices/imuBosch_BNO055/imuBosch_BNO055.cpp
```

## Pending Action

- Branch `devel_get_torque` (walking-controllers) to be PRed and merged --> @GiulioRomualdi
- Unstaged commit in YARP to be committed (into some branch) and PRed. --> whoever changed the polling delay for reading the head IMU data.

nunoguedelha, 2019-09-10:
CC @fjandrad

traversaro, 2019-09-10:
> Unstaged commit in YARP to be committed (into some branch) and PRed. --> whoever changed the polling delay for reading the head IMU data.

Perhaps this was due to @Nicogene ?

nunoguedelha, 2019-09-10:
> Repo is now on branch `fix/secondary-cal-ft-2bcb-cisco`.
> Going back to branch `devel_iCubGenova04` after merge of [robotology/robots-configuration#118](https://github.com/robotology/robots-configuration/pull/118).

PR merged. `robots-configuration` is now in `devel_iCubGenova04`. Integration branch is going to be merged into `devel` by the end of the week.

I remind you all that the integration branch `devel_iCubGenova04` is protected.

GiulioRomualdi, 2019-09-10:
> * Branch `devel_get_torque` (walking-controllers) to be PRed and merged --> @GiulioRomualdi

The `devel_get_torque` is on top of `devel`. I switched back in `devel` since implements only a way for retrieving the measured joint torque in the robot.

For the time being, `devel` branch is the stable branch used for the demos

nunoguedelha, 2019-09-10:
Thx @GiulioRomualdi

Nicogene, 2019-09-16:
> Perhaps this was due to @Nicogene ?

Nope, I worked only on the `iCubGenova02`

nunoguedelha, 2019-10-01:
> ## Pending Action
> 
>     * Branch `devel_get_torque` (walking-controllers) to be PRed and merged --> @GiulioRomualdi
=> DISCARDED (not required anymore, reverted back to `devel` branch).

> 
>     * Unstaged commit in YARP to be committed (into some branch) and PRed. --> whoever changed the polling delay for reading the head IMU data.
=> DISCARDED (as per @fjandrad this change was done only for testing purposes).

Closing this issue since all pending actions are completed.

closed 2019-10-01 by nunoguedelha