robotology/mesh-support #857: CAD feet iCub

lpenco, 2019-09-17:
#198 # Description of the failure
Is it possible to have a CAD of the feet of the iCub? of the Ft sensors and the alluminium plate of the soles?

### Detailed conditions and logs of the failure

julijenv, 2019-09-18:
Sorry for the late reply, I started yesterday convinced that I puch the button COMMENT but actually i didnt.
what are exactely the information you are looking for? could you send me a picture of the sole h=you have (the feet actually) I will try to sen you the  dxf file to look at the dimension of the CAD design. do you have a dxf software or a CAD software? let me know like that I will send you the information you need

lpenco, 2019-09-19:
[feet.zip](https://github.com/robotology/icub-tech-support/files/3630221/feet.zip)

Here a pciture. The feet are roughly 19x9cm. a dxf would be good, thx!

julijenv, 2019-09-19:
[feet_dxf_others.zip](https://github.com/robotology/icub-tech-support/files/3632389/feet_dxf_others.zip)

eccolo

lpenco, 2019-09-20:
grazie! :+1:

closed 2019-09-20 by lpenco