robotology/mesh-support #861: Right arm elbow cover broken

francescolaneve, 2019-09-19:
### Description of the failure
During an experiment for the thesis, while trying to reach home position the arm ended up in contact with the table leading to breakage of the right arm elbow cover.

### Detailed conditions and logs of the failure
I attach a picture showing the cover.blob:
[image]
[image]

julijenv, 2019-09-23:
@xEnVrE , @francescolaneve , which robot cover is broken R1 or R1-mk2?

xEnVrE, 2019-09-23:
@julijenv R1-mk2

julijenv, 2019-09-27:
@salvi-mattia ,can you please give some feedback of the work already done on this matter.
thx

salvi-mattia, 2019-09-27:
hi all,
I spoke to maggia, we decided to order a new couple of these covers at MMWS and later paint them.

Togheter with julien we noticed there are some screws to be tightened

salvi-mattia, 2019-09-30:
the parts have been ordered to mmws.
estimate delivery date: 18-oct-2019 from mmws
action to do at reception:
- painting
- assembling

pattacini, 2019-10-03:
@vtikha @maggia80 have a look at one comment up 👀 
We're a bit tight then with the mission to Milan with R1-mk2.
Is there anything we can do about that?

vtikha, 2019-10-03:
Plan B: in case it does not come through, can we "borrow" it from R1-mk1?

julijenv, 2019-10-03:
@vtikha , you mean borrow from R1, no?

vtikha, 2019-10-03:
@julijenv yes, take it from R1-mk1 just for the event.

julijenv, 2019-10-03:
when will the event take place? @vtikha

salvi-mattia, 2019-10-03:
there are some ready in our warehouse.
they are for the two R1 we will assemble in the future
if it's urgent we can borrow them

[image]

vtikha, 2019-10-03:
@julijenv 23-24 October

julijenv, 2019-10-03:
@Fabrizio69 , puoi dare il tuo ok per prendere il pezzo a magazzino?

salvi-mattia, 2019-10-11:
i componenti richiesti in officina sono pronti
per essere utilizzabili devono essere verniciati
capiamo se vogliamo mettere questi o farcene anticipare uno da @Fabrizio69 e poi rimpiazzarlo a magazzino con uno di questi.
IMHO sarebbe meglio tenere sui nostri quelli prototipati e verniciati e tenerci quelli di ARRK per i robot che "vendiamo"

vtikha, 2019-10-11:
@salvi-mattia Abbiamo preso il componente ARRK da magazzino per montarlo per l'evento, per poi restituirlo. Intanto, OK per portare a verniciare quello stampato cosi quando ritorniamo siamo sicuri di avere un pezzo da montare su `R1-mk1`.

salvi-mattia, 2019-10-14:
@maggia80 dunque procedo e porto i pezzi a verniciare

Fabrizio69, 2019-10-14:
Provvedo a scaricare 1 pezzo cod.11912 - CER_021_P_033 assegnandolo al maintenance dell'R1 S/N:001

maggia80, 2019-10-14:
@salvi-mattia si procedi se c'era già l'ordine.

salvi-mattia, 2019-10-22:
mi è arrivata la quotazione
aperta issue su purchase (spero sia giusta)
https://github.com/icub-tech-iit/purchases/issues/54
@maggia80 dai un occhio se ritieni di approvarla

salvi-mattia, 2019-11-13:
sono tornate le cover verniciate
le consegno a @julijenv 
andrebbe montata questa sul robot e restituita quella attualmente montata a michelini

julijenv, 2019-11-27:
E stato cambiato con pezzi fatto stampare e dipinto via @salvi-mattia .
i pezzi di qualita piu scrsa sono tenuti da @salvi-mattia .
I pezzi di buona fattura sono state ridate a warehouse @simonemichelini

closed 2019-11-27 by julijenv

pattacini, 2019-11-27:
cc @Miche19.