robotology/mesh-support #862: iCubGenova04: Face expressions are not working

FabioBergonti, 2019-09-30:
### Description of the failure
With @julijenv we tried to launch the face expressions several time.
We didn't succeed and we've never encountered in software errors. It seems a hardware problem

julijenv, 2019-10-09:
the teeth of the crimp for the power connector of the facial expression board were loose.

GiulioRomualdi, 2019-10-14:
Today we tried again to use the face expression, however they are not working again

S-Dafarra, 2019-10-22:
During the maker fair, this issue occurred several times and it was necessary to plug and unplug the USB connector connecting the face expressions board to the head. By checking ``dmesg`` the following error was appearing:
```
Failed to set dtr/rts.
```

julijenv, 2019-10-25:
it has been replaced main board adn recrimp the entire wires of the 2 connectors.
However, the failure keep spaming so i guess we wil need to go to the source ofthe problem by checking the connector at teh level of the icub-hesd bosrd

S-Dafarra, 2019-10-29:
Unfortunately this issue is still occurring, even after unmounting the head boards.

julijenv, 2019-10-29:
It has been unmounted entirely the main CPU boards and a visual inspection and plugging test have been performed to detect any possible flaky wires that could bring this phenomenon. Unfortunately Nothing to Report. It is highly possible that one of the USB port is damaged. (both the xsens of the bottom of your robot (which is reported failing sometimes) and the face expression are attached on the same connector. Is this related?!? It would be interesting to test this hypothesis?
let me know

S-Dafarra, 2020-02-10:
I think this has not been solved yet, but lately, we had to run them fewer times.

violadelbono, 2020-06-09:
Hello guys,

We tried to connect and disconnect the facial expressions multiple times. `dmesg` did not show any error but at some point the facial expressions stopped working.
Thus we investigated again the power connector and it is possible that the pins were not making contact because deformed or loose. Apparently this is not the first time that that type of connector is faulty.

We suggest you to try at least 5 times to connect the facial expressions and let us know if the problem still occurs.

traversaro, 2020-07-17:
Hi @violadelbono , if you need an answer from someone, it is typically safer to explicitly mention them, i.e. @GiulioRomualdi @S-Dafarra .

S-Dafarra, 2020-07-17:
Thanks @traversaro for reminding me of this. Actually, we lately we had to launch the face expressions and they did not run. So I unplugged and plugged back the connector. Next time I will try to restart the device at least 5 times as suggested.

S-Dafarra, 2020-09-20:
This is still happening. Also, independently from the number of times I try to launch the device, they won't start until we unplug and plug the connector

julijenv, 2020-10-19:
Hi @S-Dafarra ,
I think that this time around we finally make it. We recrimp every single crimps and repalce all the caoonectors.
At the first start, the face expression where recognized on the USB port with TTY conversion.
I hope this meeans that it will work for a long time now, BUT as we launched it just once, I leave this issue open until you double check by your own this miracle.
Please feel free to close the issue if ever at the first 3 to 5 next start the face expressions show up.

S-Dafarra, 2020-10-19:
> Hi @S-Dafarra ,
> I think that this time around we finally make it. We recrimp every single crimps and repalce all the caoonectors.
> At the first start, the face expression where recognized on the USB [...]

Sounds good to me, thanks a lot @julijenv @violadelbono

S-Dafarra, 2020-10-21:
Several people tested it (@gabrielenava @lrapetti @violadelbono) and it seems to be working :tada: :tada: Thanks again @julijenv @violadelbono

S-Dafarra, 2021-02-16:
Really sorry @julijenv @violadelbono for doing this to you, but this problem has come back :scream:

S-Dafarra, 2021-02-16:
I have opened https://github.com/robotology/icub-tech-support/issues/1083 as a follow-up. Closing this.

closed 2021-02-16 by S-Dafarra