robotology/mesh-support #863: iCubGenova02: shoulder cable broken

marco-monforte, 2019-10-01:
### Description of the failure

Ciao Julien,
I'm sorry but I just broke the purple iCub shoulder cable moving the joint close to one of its limits throught the motorgui.

[image]

[image]

julijenv, 2019-10-01:
@marco-monforte , are they borken and just popped-out?

marco-monforte, 2019-10-01:
Broken. Sorry, I didn't take the other ending part of the cable in the pics.

julijenv, 2019-10-01:
It has bben changed the cable N1 of the shouder.

IITCODE Qta Description
4758 x 1 : Shoulder rotation cable N1

The robot has been tested and the funtionnality restored,

closed 2019-10-01 by julijenv