robotology/mesh-support #864: ethernet link resetting when head is moving

arrenglover, 2019-10-01:
### Description of the failure

The icub-zynq ethernet connection resets under head motion.

### Detailed conditions 

I start the robotinterface and run an application called `head_control` that moves the head pitch and yaw randomly. At the same time I am logged into the `icub-zynq` (`ssh icub-zynq`). and use the `dmesg` command. I get the following output:
```
[  734.805945] macb e000b000.ethernet eth0: link down
[  736.853998] macb e000b000.ethernet eth0: link up (1000/Full)
[  740.949974] macb e000b000.ethernet eth0: link down
[  741.974028] macb e000b000.ethernet eth0: link up (1000/Full)
[  774.742170] macb e000b000.ethernet eth0: link down
[  776.790241] macb e000b000.ethernet eth0: link up (1000/Full)
[  779.862237] macb e000b000.ethernet eth0: link down
[  781.910290] macb e000b000.ethernet eth0: link up (1000/Full)
[  798.294342] macb e000b000.ethernet eth0: link down
[  800.342450] macb e000b000.ethernet eth0: link up (1000/Full)
```

Can we check the physical connection of the ethernet link of the ZCB in the robot head?

### Add log files of the failure

arrenglover, 2019-10-08:
@julijenv can we have a look at this please? It is blocking us from using the robot motion and cameras at the same time. I opened it 7 days ago but didn't hear anything - is there some new protocol I should be following?
these guys should be made aware of the issue too (@aikolina @vvasco @Iaxama @marco-monforte)

julijenv, 2019-10-08:
Hi @arrenglover , sorry I have been passing quite often to your robt to fix the face and people wer working on it, so I was like i didnt show up again. my bad.
I would nee you tomorrow to check that failure, when r u around??
let me know

arrenglover, 2019-10-08:
@julijenv - I'm working on the robot from 9am - 2pm tomorrow. Let me check the problem still happens by myself to not waste your time (something might have changed if people have been working on the robot). I'll add a message here if I need your help.

arrenglover, 2019-10-09:
@julijenv - thanks for coming this morning. I did have some troubles again today however, I'm not sure if it was from the ethernet connection, as first thought, or some problem with the ZCB. I've talked with @marco-monforte and @Iaxama to make sure we investigate exactly where the problem is coming from.
@ChiaraBartolozzi - this is the bug I was talking about. I'm not sure if it's ethernet related, or something in the HPU driver causing the Kernel to pause for a moment and the ethernet to drop. We'll keep investigating

aikolina, 2019-10-18:
I switched off the event-cameras and tried to move the neck yaw and got the same error as @arrenglover (_i.e._ link up/down of eth0).

Not sure if this information is going to be useful, but I noticed that the error was happening frequently when moving the neck yaw between -20 and +20 degrees.

I also tried to move the head in idle manually and it seemed that the neck pitch and roll were more stiff compared to neck yaw.

arrenglover, 2019-10-22:
@julijenv - I would like to do some experiments as soon as possible. Can you give me a time when we can look at this together? Thanks a lot.

julijenv, 2019-10-22:
Hi @arrenglover , if you are around the robot I ll come to look for this issue right know.

arrenglover, 2019-10-23:
@simeonedussoni @julijenv  - FYI: I had no problems all afternoon using the the temporary network cable.

Gandoo, 2019-10-24:
ciao @julijenv @arrenglover @simeonedussoni   Ho sostituito il cavo e ricrimpato il connettore di rete. Ci sarebbe, se possibile, far modificare la staffa (vedi immagine) in modo che ci sia un po' di spazio e che i fili del connettore di rete non tocchino il metallo.
IL cavo che si e' rotto e che e' stato sostituito non e' adeguato a fare movimenti continui, come quelli che fa il collo, infatti si e' usurato dove ci sono passaggi in continuo movimento. 
[image]
@julijenv ho preso dal magazzino un pezzo con cod.13824 dovresti fare il tickets per poterlo scaricare su commessa.
@arrenglover @julijenv  appena possibile, fare test per vedere se il problema era effettivamente quello.

julijenv, 2019-10-24:
it has been teste dand we can now confirm that the ethernet connexion is 1Gbit so everything ok.

closed 2019-10-24 by julijenv