robotology/mesh-support #868: iCubGenova04 neck pitch joint goes frequently in Hardware Fault

S-Dafarra, 2019-10-14:
## Description of the failure
When moving the neck roll joint using the motorgui, the neck pitch goes frequently in hardware fault. We noticed that there is a consistent amount of friction in one of the two motors. This problem seems to be similar to https://github.com/robotology/icub-tech-support/issues/866 but the faulty motor seems to be the other one.

## Detailed conditions
See above

## Add log files of the failure
None

@julijenv @GiulioRomualdi @prashanthr05

S-Dafarra, 2019-12-13:
For some reason, this does not seem to be happening a lot lately. I think we can close this and open a new one if it is occurring again.

closed 2019-12-13 by S-Dafarra

prashanthr05, 2019-12-13:
@S-Dafarra It did happen once (or maybe a few times) while we were testing Oculus-virtualizer retargeting for the research line evaluation demos.

Maybe @GiulioRomualdi @kouroshD could confirm this?

S-Dafarra, 2019-12-13:
If so, we can reopen it.