robotology/mesh-support #873: Right Knee encoder disconnected

MiladShafiee, 2019-10-23:
## Brief description of the request/failure
The right knee actuator gives error.
## Detailed context
During testing the push recovery the robot fell down and the right knee encoder cable disconnected.

## Accompanying material

[log_icub-head.txt](https://github.com/robotology/icub-tech-support/files/3763509/log_icub-head.txt)

julijenv, 2019-10-24:
The green wire of the AEA encoder was broken.
resoldered it saved the issue.
In the menawhile of this repair I also retighten the right foot base - the screws were very loose. @S-Dafarra maybe this cool make the robot to be kind of not stable when on both feet (neverthelessquite luckily the robot was balancing on the left foot)

S-Dafarra, 2019-10-24:
Thanks @julijenv!