robotology/mesh-support #875: Problems with dragonfly left camera

Nicogene, 2019-10-28:
## Brief description of the request/failure
The left dragonfly camera has launched with bayer 640x480 has some strange artifacts(no `camCalib` involved).
It appears a line in the middle of the image.

## Screenshot

[image]
[image]
[image]
[image]
[image]
[image]

cc @ddetommaso

julijenv, 2019-10-28:
Hi @Nicogene ,
It is the first time I see that.
Did you try without bayer ?
let me know

Nicogene, 2019-10-28:
Unfortunately the only way to run 640x480 cameras is with bayer due to the limited bandwith of firewire. I will try if to run only the left camera and I will let you know.

Nicogene, 2019-12-02:
With @xEnVrE we saw a similar problem on iCubV3. I don't remember if there was the bayer coding.

xEnVrE, 2019-12-02:
It was @ 320 x 240 but I don't know whether the bayer coding is enabled in the dragon configuration files.

xEnVrE, 2019-12-02:
> Unfortunately the only way to run 640x480 cameras is with bayer due to the limited bandwith of firewire. I will try if to run only the left camera and I will let you know.

We used cameras at 640x480 in `iCubGenova01` even without Bayer coding.

Nicogene, 2019-12-02:
> Which key should I check for, in order to confirm?

It depends by the ini file used for launching the cameras, see for example :

https://github.com/robotology/robots-configuration/tree/master/iCubGenova01/camera

> We used cameras at 640x480 in iCubGenova01 even without Bayer coding.

Yeah, I did the commit, the trick was reducing the framerate to 25 fps instead 30, for bandwith issued, see https://github.com/robotology/robots-configuration/commit/2423f690519b7792574b464546046a632e612b4d

xEnVrE, 2019-12-02:
It seems that during our tests iCubV3 we were using `dragonfly2_config_{left, right}.ini`, i.e. without beyer.

julijenv, 2020-08-31:
Hi @ddetommaso ,
after changing the flat cable - see issue : https://github.com/robotology/icub-tech-support/issues/926
can you tell me if the issue is still occuring? or did it solve it?
let me know

julijenv, 2020-09-28:
@ddetommaso ,
puoi chiudere l'issue se tutto a posto, grzie?

ddetommaso, 2020-09-30:
@julijenv issue risolta!

closed 2020-09-30 by ddetommaso