robotology/mesh-support #876: Bosch IMU never starts the first time

S-Dafarra, 2019-10-29:
## Brief description of the request/failure
Every time we switch on the robot head, the Bosch IMU device fails to start and we get the following error:
```
578	9.241254	ERROR	First read from the device failed, check everything is fine and retry
581	9.251476	ERROR	cannot create device <imuBosch_BNO055>
582	9.251492	ERROR	ControlBoardWrapper: error while opening subdevice
603	9.677945	ERROR	xsensmt: Sensor timeout, no sensor measurement received in the last 0.212098 seconds.
605	9.677950	ERROR	xsensmt: Sensor timeout, no sensor measurement received in the last 0.217149 seconds.
1057	21.947317	ERROR	One or more devices failed opening... see previous log messages for more info
```
To run the robot is sufficient to restart the ``yarprobotinterface``.
# Detailed context
[log_icub-head_bosch_not_starting.txt](https://github.com/robotology/icub-tech-support/files/3783943/log_icub-head_bosch_not_starting.txt)

## Accompanying material

julijenv, 2019-10-29:
Hi @S-Dafarra ,
it is a known issue, it is on the pipe of issue to be resolved, it should be fixed in the near, stay tuned

Nicogene, 2019-10-29:
This patch(https://github.com/robotology/yarp/pull/2034) introduced an error message in case the first 10 read from the sensor fail(at startup time).
Maybe increasing the number of trial to ~50 (50x0.01ms) could fix the problem.
I suspect that once you turn on the icub-head the IMU takes some time to wake up.

Nicogene, 2019-10-29:
I want to underline that probably it is a problem that exists since forever, simply before robotology/yarp#2034 there was that other problem robotology/yarp#1994.

BTW the imubosch mounted on the rfe board seems to not have this issue and it is ready to be used(https://github.com/robotology/icub-main/issues/570).

Nicogene, 2019-10-29:
I tried on `iCubGenova02` with 1000 trials, and the read on i2c doesn't fail, but it read garbage, after the power on this condition is always false:

``` c++
      double sum_squared = quaternion_tmp.w() * quaternion_tmp.w() +
                             quaternion_tmp.x() * quaternion_tmp.x() +
                             quaternion_tmp.y() * quaternion_tmp.y() +
                             quaternion_tmp.z() * quaternion_tmp.z();

        if( (sum_squared < 0.9) || (sum_squared > 1.2) )
        {
            dataIsValid = false;
            return;
        }
``` 
I don't remember the rationale behind this check.

S-Dafarra, 2019-10-29:
I think it is checking the quaternion to have unit norm :thinking:

julijenv, 2019-11-04:
Would it be possible to find the reason why it needs 2 tarts to have the condition accpeted? or how long it takes to have the condition accpeted?

Nicogene, 2019-11-04:
The reason is this https://github.com/robotology/yarp/issues/1994.

Removing that check, at the first startup the imu doesn't work, it gives only `0.0` and `nan`.

pattacini, 2020-01-03:
@Nicogene @julijenv 
What's the status of this?

Is the problem related to software or hardware?
Is the RFE available on iCubGenova04?

Nicogene, 2020-01-03:
I honestly don't know how to approach this issue :thinking: 

In my opinion is not something software related, but I don't know how to check it.
The imu on the rfe should not have this kind of problem, but unfortunately `iCubGenova04` doesn't have it.

traversaro, 2020-01-03:
> I honestly don't know how to approach this issue 🤔

I just noticed that in the error log posted by @S-Dafarra there is also a related error from the `xsensmt` driver (see https://github.com/robotology/xsensmt-yarp-driver/blob/6c63c90ebb15dc42dc78a44ff5abef3051e8b18a/xsensmt/XsensMT.cpp#L71) : 
~~~
603	9.677945	ERROR	xsensmt: Sensor timeout, no sensor measurement received in the last 0.212098 seconds.
605	9.677950	ERROR	xsensmt: Sensor timeout, no sensor measurement received in the last 0.217149 seconds.
~~~
I wonder if the two issues may be related, even if the device are attached to the robot using two different bus. 

If we want to investigate the issue, I would try the following: 
* Detach the `xsensmt` waist IMU and see if the issue with the `bno055` persists.  
* Instead of using the [`udev`-assigned names](https://www.freedesktop.org/software/systemd/man/udev.html) such as [`/dev/icub-i2c-imu`](https://github.com/robotology/icub-tech-support/issues/763#issuecomment-457628681) in https://github.com/robotology/robots-configuration/blob/8464607c26ce4bcefad5f85a72e84a76e026cbff/iCubGenova04/hardware/inertials/head-inertial.xml#L6 and try to use the raw device, and see if the problem persists. 
* Try to connect to the sensor on the first trial (when it fails to open via YARP) using a different driver, for example https://github.com/evdrvn/bno055-i2c or https://github.com/ghirlekar/bno055-python-i2c, and see if the issue persist also in that case.

S-Dafarra, 2020-03-03:
This has not been fixed yet and the suggestions in https://github.com/robotology/icub-tech-support/issues/876#issuecomment-570580196 have not been tried AFAIK.

Nicogene, 2020-03-04:
A workaround could be do the restart of the device inside of the code of `imuBosch_BNO055` in case of failure of startup, in order to avoid to re-run the `yarprobotinterface`

traversaro, 2020-05-03:
Hi good friend @stale, I am afraid that this issue is still present.

julijenv, 2020-07-01:
@Nicogene , do u have any other good ideas to make it ok?
https://github.com/icub-tech-iit/fix/issues/145

julijenv, 2020-07-01:
A bug fix has been requested on icub-fix channel and will be addressed when FW guy available
icub-tech-iit/fix#145

Nicogene, 2020-07-01:
> @Nicogene , do u have any other good ideas to make it ok?

Unfortunately not, but I'd try what @traversaro suggested here -> https://github.com/robotology/icub-tech-support/issues/876#issuecomment-570580196

DanielePucci, 2021-06-07:
Hi all, unfortunately we keep having this issue. So, I am going to re-open it. Let un know how you prefer to proceed 

CC
@maggia80

Uboldi80, 2022-01-25:
a feasibility study was recently carried out to introduce RFE master on the robot head, which should help solve this problem as well as the problem of face expressions.

closed 2023-02-21 by github-actions[bot]