robotology/mesh-support #877: How to use the bluetooth battery indicator

S-Dafarra, 2019-10-29:
## Brief description of the request/failure
During demos in battery power, we usually experience [range anxiety](https://en.wikipedia.org/wiki/Range_anxiety) leading us to press the button on the battery every few minutes. Nonetheless, we made the robot fall a couple of times because we ran out of battery. I remember a discussion with @maggia80 (and probably @MrAndrea) regarding a Bluetooth battery indicator, but I don't know how to use it. Is there any entry point for this?

## Detailed context

## Accompanying material

cc @GiulioRomualdi

maggia80, 2019-10-29:
hi @S-Dafarra, there is a piece of code to see the status of the battery. @randaz81 did it years ago.
@randaz81 could you please give us hints on where to find it and eventually how to use it?
iCubHead with the COMExpress has the bluetooth pci-express board configured. It should be easy to run the app on the head.

S-Dafarra, 2019-12-29:
Hi @randaz81, do you remember how to run the piece of code mentioned by @maggia80 by any chance?

closed 2020-03-05 by stale[bot]