robotology/mesh-support #878: Icub initialization problem--no reader from shoulder encoder

lx2007, 2019-10-30:
We are the new user of Icub, when we do the initialization, we find the right arm can not move. After checking by Icub doctor Julien, it seems there is no signal for one encoder. So, how can we solve this problem?

lx2007, 2019-10-30:
[image]
[image]

Follow the instruction, we tried to use multi-meter to put the black probe on the red cable, and red probe to other cables one by one, however, there is no Bip sound. So, what should we do next? thanks

julijenv, 2019-10-30:
Hi @lx2007 , during our recent temviewer session and skype (wechat), we discover that the yarprobotinterface was reporting the following message

> right-arm (j0-3) - AEA encoder reading ERROR j1.

We therefore ask the user to open the upper ar mcover to investigate the sensor.

> as described over skype this morning please check the integrity of the cable labelled : R0S1 – it may have been damaged or stretched on the travel or when unpacked.

[image]

Not being able to see broekn wire, I ask to make a simple diode test in between pin of the sensor.

> Please use a multimeter in diode mode to detect the wire that failed. 
Put the Black Probe on the Red wire (pin 1 – most left on the picture above)
Red probe move from white cable to little black wire, for all of them you should be hearing a bip nose to confirm that it works. If you don’t hear anything for wire white-yellow-green most probably the failure wire is red or little black (the big black is just the shiedl).

lx2007, 2019-10-30:
Unfortunately, i did not hear any sound even i tested both ends of the junction. Here is my multimeter setting 
[image]

julijenv, 2019-12-18:
After my visit in Shenzhen, here the result and explanation of what has been done on the robot to have it reborn.
The board EMS11 from the skin of the right leg has been swapped with the one from the EMS3 from the right shoulder (the SPI line of PB1 was not functionning anymore).
It has been later on performed the full exit-test to make sure that the robot behave as when went to China.

closed 2019-12-18 by julijenv