robotology/mesh-support #879: CAD support shoulder

amaroyo, 2019-11-01:
Hi @julijenv, 
cc @reafrancesco 

I am adding this issue as a reminder to solve the problem we have with the shoulders and the cable that eventually falls into the separation of the joint. Have you managed to find the CAD so to get the dimension of the separation? 

Thanks, 
Alex.

[image]
[image]

julijenv, 2019-11-04:
Hi @amaroyo , I tried to revert CAD Design to before the old shoulder piece (the one still on your robot) to get the size of this spacer. I was not able to find out the old design. not anymore in repository svn (too old). WCS I will have to compare between both shoulder of yours and shrink down the left one if found different.

julijenv, 2019-11-05:
Hi @amaroyo, it has been checked and the piece is exactely the same in between the new shoulder and the old one (yours) so I will check what is precisely the probelm with your shoulder next time I come.

amaroyo, 2019-11-05:
@julijenv, perfect! thanks!

julijenv, 2019-11-08:
if after a week or so of testing you see that everything is ok, please close the issue thx.
@amaroyo

amaroyo, 2019-11-16:
@julijenv, the cable (nº8) of the right shoulder is back in the hole again (see image). With the current tempo of experiments, in less than a week will be broken. 

[image]

julijenv, 2019-11-21:
it has been solved, the spacer in between the shoulder and the pulley was out of specifics (1.65 instead of 1.5mm), file it make the trick - it has been done on both arm.

|IITCODE | Alias| REv.|Qty| Type|DEscription|
| --- | --- | --- | --- | --- | --- |
|4765 | RC_IIT_019_G_008 | 2 | 2  | Assemblies | SHOULDER ROTATION CABLE 8|
|4762 | RC_IIT_019_G_005 | 1 | 2  | Assemblies | SHOULDER ELEVATION CABLE 5|
|1895 | RC_IIT_003_P_006_00 |   |  2 | Freeform custom plastic parts (3D printing, molding) | LEFT PULLEY-CRIMP COVER|
|1896 | RC_IIT_003_P_007_00 |   |  2 | Freeform custom plastic parts (3D printing, molding) | RIGHT PULLEY-CRIMP COVER|

closed 2019-11-21 by julijenv