robotology/mesh-support #880: right shoulder cables about to break

amaroyo, 2019-11-01:
Hi @julijenv, 

Right shoulder, I think they are the cables 1 and 2. It's strange that they are wearing down as there's no direct friction with anything. Check the images attached - also one of those cables is getting twisted, similarly to the one you changed some time ago. 

Cheers, 
Alex.

[image]
[image]
[image]

amaroyo, 2019-11-04:
Adding to the issues from before, right thumb distal and proximal broken.

julijenv, 2019-11-04:
I think then, it is another issue to open?!?!
@amaroyo  :P

amaroyo, 2019-11-05:
@julijenv, our robot is falling apart... they tried to fix the shoulder today and this plastic part broke...

I think it will be convinient to come over half a day to fix all of them, and take a look to the CAD issue. 

[image]

julijenv, 2019-11-05:
Ok I will let you know tomorrow. when you would need to pass by and give me a lift.
by the way this piece is not really importatn

closed 2019-11-08 by julijenv