robotology/mesh-support #881: iCubErzelli01: CAD models

ddetommaso, 2019-11-04:
## Brief description of the request/failure
I'd like to know where I can find the CAD (Inventor?) models of the iCub's head with detail of the boards/motors etc.. Specifically I'd need the one of the iCubErzelli01.

julijenv, 2019-11-05:
@Mick3Lozzo , could you try to send me a CAD of the head in pdf 3D., thx in advance

ddetommaso, 2019-11-05:
@julijenv would it be possible to have also models like the ones in this repo?

https://sourceforge.net/p/robotcub/code/HEAD/tree/trunk/iCubHardware/mechanics/

Mick3Lozzo, 2019-11-05:
[IC_002_A_001_STDHEAD.pdf](https://istitutoitalianotecnologia-my.sharepoint.com/personal/michele_gesino_iit_it/Documents/ic_002_a_001_stdhead.pdf?web=1)

This is a link with a 3D pdf of the head. 

@ddetommaso The whole repository has been moved to GitHub, at the following address:

https://github.com/icub-tech-iit/cad-mechanics.git

the iCub subfolder is 

https://github.com/icub-tech-iit/cad-mechanics/tree/master/projects/icub/icub1_icub2

Mick3Lozzo, 2019-11-05:
> @julijenv would it be possible to have also models Creo (Pro/ENGINEER) like the ones in this repo?
> 
> https://sourceforge.net/p/robotcub/code/HEAD/tree/trunk/iCubHardware/mechanics/

This is an old and outdated version of the repository. Don't consider it.

ddetommaso, 2019-11-05:
> [IC_002_A_001_STDHEAD.pdf](https://istitutoitalianotecnologia-my.sharepoint.com/personal/michele_gesino_iit_it/Documents/ic_002_a_001_stdhead.pdf?web=1)
>
> This is a link with a 3D pdf of the head. [...]

Thanks a lot @Mick3Lozzo 
It seems I do not have access to the GitHub repository, could you please authorize my account?

Mick3Lozzo, 2019-11-05:
... I have not _so much_ power.

@maggia80, @pattacini?

> 
> Thanks a lot @Mick3Lozzo
> It seems I do not have access to the GitHub repository, could you please authorize my account?

pattacini, 2019-11-05:
@ddetommaso the repository won't be accessible for the moment. In the next future, we're going to assess the right modality to provide the information that can be disclosed.

As of now, you'll have to digest the file @Mick3Lozzo posted above and/or refer to the old repository on SVN.

pattacini, 2019-11-05:
Of course, in the meantime, if you need further info that you won't find in the SVN repo, just get back to us.

ddetommaso, 2019-11-05:
@pattacini I can understand the restrictions in authorization policies for private repositories, but I find more difficult to understand restrictions in accessing certain information (up-to-date) about the iCub platform. The PDF will be surely helpful and I'll reopen the issue in case we'll need further info.

Thank you

closed 2019-11-05 by ddetommaso

pattacini, 2019-11-05:
@ddetommaso the problem is that with one Git repository we don't have the fine-grain control of which subfolders we can make public as it happens with Subversion. Given that the repo cad-mechanics contains information that we need to keep confidential, we are somehow forced now to hide the whole content.

Of course, we will soon undertake the necessary actions to restore the accessibility of those data that were used to be public.

maggia80, 2019-11-08:
@ddetommaso I think that you should look at the SVN: https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/ where you can find the updated files of the head, waiting for the git repo. The sourceforge repo is very old

ddetommaso, 2019-11-08:
Thanks a lot @maggia80!

pattacini, 2019-12-28:
Hi @ddetommaso 

Here you go with our public repository containing CAD schematics: https://ieeexplore.ieee.org/document/8794435.

Please, consider this structure provisional as it still needs to be reviewed by our CAD designers.

adamlukomski, 2019-12-28:
@pattacini are you sure this is a correct link?

> Hi @ddetommaso
> 
> Here you go with our public repository containing CAD schematics: https://ieeexplore.ieee.org/document/8794435.
> 
> Please, consider this structure provisional as it still needs to be reviewed by our CAD designers.

pattacini, 2019-12-28:
Sorry, here's the correct link: https://github.com/icub-tech-iit/cad-mechanics-public